wpilibsuite/allwpilib #4899: 2023.1.1 Intel Mac install might be corrupt

swalberg, 2023-01-04:
**Describe the bug**
I can't install the 2023.1.1 Intel Mac because I'm told the artifacts file was damaged

**To Reproduce**
Steps to reproduce the behavior:
1. Download the file from the releases page
2. Attempt to install. Observe error after it attempts to verify the artifacts file

**Expected behavior**
I expect to be able to install the new version.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1
 - OS: OSX 13.1
 
**Additional context**
I re-downloaded the file in case my download was corrupt, though I was able to open the .dmg and pass the OSX inspection.

```sean/Volumes/WPILibInstaller 2$ shasum WPILib_Mac-2023.1.1-artifacts.tar.gz
b32d4eebd66c74a9a933d5a32d2a8464383bc8c3  WPILib_Mac-2023.1.1-artifacts.tar.gz
sean/Volumes/WPILibInstaller 2$```

ThadHouse, 2023-01-04:
Just to double check, you’re not attempting to install the intel installer on an arm Mac, right? That’s the error that occurs in that case.

swalberg, 2023-01-04:
Nope, I'm on an Intel Mac and am installing the Intel version.

ThadHouse, 2023-01-04:
Ok. We’ll try to get someone with an intel Mac to look. Most of our devs have upgrade to arm.

ThadHouse, 2023-01-04:
Just to double check, can you go into the app bundle (right click and explore contents), find the checksum.txt file in there and post the contents of that?

swalberg, 2023-01-04:
It's working now, this one is on me. I originally opened the ARM version and it wouldn't let me start because I'm on an Intel box so I closed that and tried again with Intel. I just deleted and unmounted everything, and after redownloading the Intel version it's working. Not sure how the Intel installer tried to look at the ARM artifacts 🤷

closed 2023-01-04 by swalberg

ThadHouse, 2023-01-04:
The way we have to find the artifacts is a bit wonky, so if you have 2 different installer disk images mounted things get weird. And likely just found that resource first.