wpilibsuite/allwpilib #4900: [nt4] Glass repeatedly gets disconnected then reconnected on NT4

ohowe1, 2023-01-04:
**Describe the bug**
When using glass (and outline viewer) with a rio, the connection will repeatedly reset. Seen on two computers with different OSs. Using most recent beta rio image. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open Glass
2. Connect to robot

**Expected behavior**
It to stay connected

**Screenshots**
In the rio log:
```
NT: DISCONNECTED NT4 client 'glass' (from 10.37.29.231:49358): 
NT: Got a NT4 connection from 10.37.29.231 port 49362
NT: Got a NT4 connection from 10.37.29.231 port 49365
NT: CONNECTED NT4 client 'glass' (from 10.37.29.231:49362)
NT: CONNECTED NT4 client 'glass@1' (from 10.37.29.231:49365)
NT: DISCONNECTED NT4 client 'glass' (from 10.37.29.231:49362): stream error: ECONNRESET
NT: NT4 socket error: socket is not connected
NT: NT4 socket error: connection reset by peer
```

In the glass log
```
CONNECTED NT4 to 10.37.29.2 port 5810 (NetworkClient.cpp:421)
DISCONNECTED NT4 connection:  (NetworkClient.cpp:455)
```
Both of these are repeated in the console

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1-beta-7
 - OS: macOS & Windows 10
 - Java version: 17

sciencewhiz, 2023-01-04:
Does this occur with a simple robot project (ie on of the examples)?

ohowe1, 2023-01-04:
I can’t reproduce on a template project with some data being sent over NT.

ohowe1, 2023-01-04:
But weirdly enough when I load my other project back in the issue is no longer there…

superpenguin612, 2023-01-31:
I am having the same issue, on two separate computers connected to the same robot. It likely has something to do with the Field2d object.

Starlight220, 2023-01-31:
@superpenguin612 based on what are you suspecting that it's relevant to Field2d?

I think we've had reports of these NT disconnects in SysId as well, so it seems unlikely that the cause is Field2d.

PeterJohnson, 2023-02-04:
Hopefully fixed by #5048.

PeterJohnson, 2023-02-06:
Please see if 2023.3.2 fixes the issue for you.  Feel free to reopen if it doesn't.

closed 2023-02-06 by PeterJohnson