wpilibsuite/allwpilib #4905: getResourceAsStream always returns null in desktop simulator.

blu28, 2023-01-05:
**Describe the bug**
I have some files that are stored under src/main.resources. At runtime, I should be able to read those files using the getResourceAsStream method to open a buffered stream to them. This worked fine last year. However, this year, when I use the simulator, it always returns null. The line of code looks like this:

InputStream statusfile = Main.class.getResourcesAsStream("/file.txt");

Looking at the project directory, I see that in the past the project jar file that is built in builds/lib was copied to the project root directory, but when running the simulator it no longer is copied. Manually copying the project jar file to the project root allows the program to run and not throw a null exception, but of course that is not an acceptable solution. 

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1
 - OS: Windows 10
 - Java version 17.0.5

blu28, 2023-01-05:
While trying to find a workaround, I was debugging and stepping through the code and it suddenly started working and even after everting all my source code changes it continued to work. I even deleted the project and cloned it again from github and it still works.

Starlight220, 2023-01-22:
Can this be closed?

closed 2023-01-23 by PeterJohnson

blu28, 2023-01-28:
It happened again on a newly cloned repo. There is something going on but something fixes it. If it happens again I will try to figure it out.