wpilibsuite/allwpilib #4908: File picker doesn't open when GTK_USE_PORTAL=1

Etaash-mathamsetty, 2023-01-06:
**Describe the bug**
look at the title

**To Reproduce**
Steps to reproduce the behavior:
1. try to open a file in wpilib vscode

**Expected behavior**
the system file picker should open 

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1
 - OS: Arch linux KDE

It works perfectly on the normal vscode

Etaash-mathamsetty, 2023-01-06:
ok I fixed it by using the system vscode and installing wpilib extensions in it (I have to do further testing to make sure that this solution works though)

sciencewhiz, 2023-01-06:
Can you post the contents of the vscode about window for both the system vscode and the wpilib one?

Etaash-mathamsetty, 2023-01-06:
system one: 
```
Version: 1.74.2
Commit: e8a3071ea4344d9d48ef8a4df2c097372b0c5161
Date: 2022-12-20T10:27:19.253Z
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 6.1.1-zen1-1-zen
Sandboxed: No
```
wpi:
```
Version: 1.74.2
Commit: e8a3071ea4344d9d48ef8a4df2c097372b0c5161
Date: 2022-12-20T10:27:19.253Z
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 6.1.1-zen1-1-zen
Sandboxed: No
```
and now it decided to start working for absolutely no reason, man I love software sometimes