wpilibsuite/allwpilib #4921: Unable to add NT entries from OutlineViewer

fovea1959, 2023-01-09:
**Describe the bug**
2023 Outline Viewer does not correctly add new values to network tables.

**To Reproduce**
Steps to reproduce the behavior:
1. Put together an empty java robot project from a 2023 template. Add one line to robotInit: `SmartDashboard.putBoolean("fovea1959", false);`
2. Simulate the robot.
3. Start 2023 OutlineViewer. set it up to be a client for 'localhost', either NT3 or NT4.
4. Observe that the 'SmartDashboard/fovea1959' entry is present.
5. Right click on the 'SmartDashboard' node in OutlineViewer, and select "Add new...". Type a new name, and select any data type.
6. Notice that the entry is created, but has no type:
[image]
7. Start 2022 OutlineViewer. set it up to be a client for 'localhost.
8. Observe that the 'SmartDashboard/fovea1959' entry is present.
9. Right click on the 'SmartDashboard' node in 2022 OutlineViewer, and select "Add new...". Type a new name, and select any data type, fill in the value.
10. Notice that the entry is created correctly:
[image]

**Expected behavior**
I expect the network table entry to be added with a type and the ability to change it's value.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1
 - OS: Windows 10
 - Java version whatever shipped with WPILIB 2023.1.1
 - C++ version N/A

**Additional context**
Add any other context about the problem here.

sciencewhiz, 2023-01-23:
Ran into this recently also

webbbt456, 2023-02-07:
Replacing the CreateTopicMenuItem function in the file glass/src/libnt/native/cpp/NetworkTables.cpp (approximately line 1040) with the following code, fixes most of the problems.  This works for both glass and outlineviewer. 

Entering float arrays does not work in either (double array do).
Adding of types with the exception of double and float work.
If you have glass as a client and outlineviewer as a server, the entries get added on both.  The value appears in the app it was created in, but does not appear on the other app. Once the value is changed in the app it was created in, the value propagates to the other app.
Notice for float and double topics the line of code is different than the other types.  The line that is commented out always crashes either program. The line of code that is not commented out may cause the app to crash.

 
#include <networktables/BooleanTopic.h>
#include <networktables/IntegerTopic.h>
#include <networktables/FloatTopic.h>
#include <networktables/DoubleTopic.h>
#include <networktables/StringTopic.h>
#include <networktables/BooleanArrayTopic.h>
#include <networktables/IntegerArrayTopic.h>
#include <networktables/FloatArrayTopic.h>
#include <networktables/DoubleArrayTopic.h>
#include <networktables/StringArrayTopic.h>
#include <networktables/GenericEntry.h>
static void CreateTopicMenuItem(NetworkTablesModel* model,
                                std::string_view path, NT_Type type,
                                const char* typeStr, bool enabled) {
  if (ImGui::MenuItem(typeStr, nullptr, false, enabled)) {
    NT_Topic ntTopic(nt::GetTopic(model->GetInstance().GetHandle(), path));
    auto entry = model->AddEntry(ntTopic);
    if (entry->publisher == 0) {
      entry->publisher = nt::Publish(entry->info.topic, type, typeStr);
      entry->persistent = path.starts_with("/Preferences/");
      nt::Topic(ntTopic).SetPersistent(entry->persistent);
      switch (type) {
      case NT_BOOLEAN:
        model->GetInstance().GetBooleanTopic(path).Publish().Set(false);
        break;
      case NT_INTEGER:
        model->GetInstance().GetIntegerTopic(path).Publish().Set(0);
        break;
      case NT_FLOAT:
        nt::Topic(ntTopic).GenericPublish(typeStr).SetFloat(0.0f);  // May crash program
        //model->GetInstance().GetFloatTopic(path).Publish().Set(0.0f);  // CRASHES Program
        break;
      case NT_DOUBLE:
        nt::Topic(ntTopic).GenericPublish(typeStr).SetDouble(0.0);  // May crash program
        //model->GetInstance().GetDoubleTopic(path).Publish().Set(0.0);  // CRASHES Program
        break;
      case NT_STRING:
        model->GetInstance().GetStringTopic(path).Publish().Set("");
        break;
      case NT_BOOLEAN_ARRAY:
        model->GetInstance().GetBooleanArrayTopic(path).Publish().Set(std::vector<int>());
        break;
      case NT_INTEGER_ARRAY:
        model->GetInstance().GetIntegerArrayTopic(path).Publish().Set(std::vector<std::int64_t>());
        break;
      case NT_FLOAT_ARRAY:  // Can be added, but ui entry does not work
        model->GetInstance().GetFloatArrayTopic(path).Publish().Set(std::vector<float>());
        break;
      case NT_DOUBLE_ARRAY:
        model->GetInstance().GetDoubleArrayTopic(path).Publish().Set(std::vector<double>());
        break;
      case NT_STRING_ARRAY: 
        model->GetInstance().GetStringArrayTopic(path).Publish().Set(std::vector<std::string>());
        break;
      default:
        break;
      }
    }
  }
}

closed 2023-03-20 by PeterJohnson