wpilibsuite/allwpilib #4922: WPILib Simulation Crash using NavX software

NetLockJ, 2023-01-10:
**Describe the bug**
Upon simulation in a project, code will build as expected. When doing a `WPILib: Simulate Robot Code` everything builds just fine, but after selecting the GUI and moving on, the simulation crashes and gives the following stacktrace. I don't know if this is a problem with WPILib or with Kauai Labs software, but this is a consistent problem over all cases of WPILib installs. I've tried on both Windows and macOS and it gives the same result.

Simulations without any NavX software however will build and simulate just fine.

```
HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.
Simulator GUI Initialized!
HAL Extensions: Successfully loaded extension
********** Robot program starting **********
NT: could not open persistent file 'networktables.json': No such file or directory
NT: Listening on NT3 port 1735, NT4 port 5810
navX-Sensor Java library for FRC
Error at com.kauailabs.navx.frc.SimIO.<init>(SimIO.java:59): Unhandled exception: java.lang.NoSuchMethodError: 'edu.wpi.first.hal.SimBoolean edu.wpi.first.hal.SimDevice.createBoolean(java.lang.String, boolean, boolean)'
        at com.kauailabs.navx.frc.SimIO.<init>(SimIO.java:59)
        at com.kauailabs.navx.frc.AHRS.<init>(AHRS.java:225)
        at com.kauailabs.navx.frc.AHRS.<init>(AHRS.java:357)
        at com.kauailabs.navx.frc.AHRS.<init>(AHRS.java:346)
        at frc.robot.RobotContainer.<init>(RobotContainer.java:24)
        at frc.robot.Robot.robotInit(Robot.java:40)
        at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:106)
        at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:343)
        at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$0(RobotBase.java:413)
        at java.base/java.lang.Thread.run(Thread.java:833)

Warning at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:358): The robot program quit unexpectedly. This is usually due to a code error.
  The above stacktrace can help determine where the error occurred.
  See https://wpilib.org/stacktrace for more information.
Error at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:365): The startCompetition() method (or methods called by it) should have handled the exception above.
```

**To Reproduce**
Steps to reproduce the behavior:
Run a `WPILib: Simulate Robot Code` and hit enter when selection of GUI and SmartDashboard comes up

**Desktop (please complete the following information):**
 - WPILib Version: [Latest 2023 version]
 - OS: [macOS Monterey 12.6]
 - Java version [openjdk 17.0.5 2022-10-18]

**Additional context**
If you need additional information, let me know, I'm happy to work to resolve this issue!

ThadHouse, 2023-01-10:
You need to be using the newest 2023 vendor dep for NavX. The 2022 one will give that error. The docs have the current year release in them.

NetLockJ, 2023-01-10:
Yep, that worked. I went and deleted the json files in `vendordeps` and reinstalled the libraries. Just a side note then, the check for updates didn't fix the problem when I ran that, but that could be a 2022-2023 thing. 

Thanks for the quick answer!

closed 2023-01-10 by NetLockJ

sciencewhiz, 2023-01-10:
The location of the vendordep json changed. WPILib has no way of knowing that, thus it reports that there's no updates available.