wpilibsuite/allwpilib #4933: Generalize the odometry and pose estimator classes to 3D

calcmogul, 2023-01-12:
The tilting mechanism on the 2023 field means robots are not always on the x-y plane. We should be able to generalize the odometry and pose estimator classes to handle this case though.

We can make the internals use 3D poses instead of just 2D and use Pose3d.exp(). The encoder data would go in Twist3d.dx and Twist3d.dy Twist3d.dz is zero), and the Twist3d’s orientation change would be the change in an IMU’s full 3D orientation.

We can keep the 2D gyro measurement and vision pose overloads and add 3D ones, so this wouldn't be an API breaking change, in theory.

spacey-sooty, 2024-07-28:
Should we just 0 values in 2d overloads?

calcmogul, 2024-07-28:
The options are:

1. Make the existing pose estimator use 3D internally and expose method overloads for 2D and 3D.
2. Make separate pose estimator classes for 2D and 3D.

Option 1 is easier, but has a performance cost for those who only care about 2D. With that said, any games with a driving surface that isn't entirely flat will require the 3D version for accuracy reasons. I'd use 3D all the time with the IMU's Rotation3d in my own robot code.

PR #7119 (2024-11-16): [wpimath] Add 3d odometry and pose estimation
Opening mostly just to get the code out there.

Resolves #4933.

Still needs to be timed to determine whether the performance is good enough to always use 3D or if we need to maintain a separate 2D implementation to not affect performance for the 2D case too much. Local accuracy tests show very similar results, though, on the order of 1e-12 meters and 1e-12 radians. (Local accuracy tests were done by adding some logging code to `testFollowTrajectory()` in `SwerveDrivePoseEstimatorTest.cpp`, copying into `SwerveDrivePoseEstimator3dTest.cpp`, replaci [... cut]
closed 2024-11-16 by PeterJohnson