wpilibsuite/allwpilib #4946: SequentialCommandGroup and ParallelGC class overridden methods are marked final in latest release

schleprock, 2023-01-14:
with the 2023 release, why are the overriden methods (initialize, end ...) marked as final in the SequentialCommandGroup, ParallelCommandGroup and ParallelDeadlineCommandGroup classes? for 2022 we overrode those methods to trace when these commands are placed in the queue and when they are removed. we also overrode them to setup some conditions needed for the command group. for example:

public class MyCmd extends SequentialCommandGroup {
...
@Override
public void initialize() {
  Trace.getInstance().logCommandStart(this);
  Robot.getGyro().setInitialOffset(offset);
  super.initialize();
}
...

with the methods now marked final, we cannot override them as shown above.

the question now becomes how do we call methods needed in initialize() or end() or ... now that these methods are marked final? so in the above example, how do we call logCommandStart() and setInitialOffset() when the command's initialize() method gets called?

if there is no way to do this now, then we really need a new release that removes the final keyword from those methods in the classes.

Starlight220, 2023-01-14:
Those methods were marked final because overriding them was *super* unsafe (pun is a happy accident).
If you need to override, use delegation rather than inheritance. You shouldn't need to, as there should always be a better solution.
Command-based in general is moving away from subclassing (especially from classes other than CommandBase), and towards defining commands inline with lambdas. The docs were recently updated for this; I recommend taking a look.

As for logging when commands are scheduled and such, I recommend looking at command event listeners such as `CommandScheduler.onCommandInitialize(Consumer<Command>)`.

Starlight220, 2023-01-22:
OP isn't responding, and this is a wontfix. Closing.

closed 2023-01-22 by Starlight220