wpilibsuite/allwpilib #4962: Create Timer.restart() method

srimanachanta, 2023-01-17:
Most uses for the Timer class use this snippet of code:
```
Timer timer = new Timer();
timer.start();
// do something
timer.stop();
// some restart condition
timer.reset();
timer.start();
```

There should be a `Timer.restart()` method that will reset the timer and start it again (that list bit of code).

calcmogul, 2023-01-17:
The `start()` after `reset()` is redundant. `reset()` doesn't affect whether the timer is started or stopped, so just calling `reset()` does what you want.
https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/Timer.java#L86-L89

srimanachanta, 2023-01-17:
> The `start()` after `reset()` is redundant. `reset()` doesn't affect whether the timer is started or stopped, so just calling `reset()` does what you want if you leave the timer running. https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/Timer.java#L86-L89

The application I am referring to are times when the timer is used within a command and the timer must be restarted and the time counter reset to 0.

sciencewhiz, 2023-01-17:
Why doesn't just reset work in that case?

srimanachanta, 2023-01-17:
> Why doesn't just reset work in that case?

The `end()` method of the command stops the timer, it needs to be restarted

rzblue, 2023-01-17:
Unless you're using the accumulated time across stops you don't need to call stop- the timer doesn't do anything in the background. Calling reset will reinitialize the start time to the current time.

PR #4963 (2023-01-23): [wpilib] Add Timer::Restart()
closes #4962
closed 2023-01-23 by PeterJohnson