wpilibsuite/allwpilib #4965: NetworkTables symbol lookup error on Linux x86-64

rmheuer, 2023-01-17:
**Describe the bug**
When trying to start a NetworkTables client on Linux x86-64, there is a symbol lookup error for `_ZN3wpi6detail18PromiseFactoryBase13CreateRequestEv`.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Gradle project with ntcore and wpiutil Java and JNI dependencies for linux x86-64
2. Use `NetworkTableInstance#startClient4` or `NetworkTableInstance#startClient3` to start a client
3. Run the project, the symbol lookup error will occur

**Expected behavior**
The client would start with no errors, and connect to the server if it is running.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.2.1
 - OS: Linux Mint 21 x86_64
 - Java version: OpenJDK 17.0.5
 - C++ version: No idea how to find this

**Additional context**

Log output:
```
NT: starting network client
/usr/lib/jvm/java-17-openjdk-amd64/bin/java: symbol lookup error: /home/ultraviolet/.wpilib/nativecache/linux/x86-64/libntcorejni.so.4646d455f80d2f18ea5fdcf34271bfb1: undefined symbol: _ZN3wpi6detail18PromiseFactoryBase13CreateRequestEv

```

Code used to reproduce:
```
import edu.wpi.first.networktables.NetworkTableInstance;

public class Test {
    public static void main(String[] args) {
        NetworkTableInstance instance = NetworkTableInstance.getDefault();
        instance.setServer("localhost");
        instance.startClient3("Test");
    }
}
```

build.gradle (info not relevant to the issue not included)
```
compileJava {
    sourceCompatibility = '11'
    targetCompatibility = '11'
}

def WPILIB_VERSION = '2023.2.1'

repositories {
    mavenCentral()

    maven {
        url 'https://frcmaven.wpi.edu/artifactory/release'
    }
}

dependencies {
    implementation "edu.wpi.first.ntcore:ntcore-java:${WPILIB_VERSION}"
    implementation "edu.wpi.first.ntcore:ntcore-jni:${WPILIB_VERSION}:linuxx86-64"
    implementation "edu.wpi.first.wpiutil:wpiutil-java:${WPILIB_VERSION}"
    implementation "edu.wpi.first.wpiutil:wpiutil-jni:${WPILIB_VERSION}:linuxx86-64"
}

jar {
    from {
        configurations.compileClasspath.collect { it.isDirectory() ? it : zipTree(it) }
        configurations.runtimeClasspath.collect { it.isDirectory() ? it : zipTree(it) }
    }
    duplicatesStrategy = DuplicatesStrategy.INCLUDE
}
```

PeterJohnson, 2023-01-17:
Try adding wpinet to the list of dependencies.

rmheuer, 2023-01-17:
The error still occurs with wpinet dependencies:
```
    implementation "edu.wpi.first.wpinet:wpinet-java:${WPILIB_VERSION}"
    implementation "edu.wpi.first.wpinet:wpinet-jni:${WPILIB_VERSION}:linuxx86-64"
```

ThadHouse, 2023-01-17:
The -jni classifiers no longer reliably work in 2023. They've always been problematic, and some of the changes made this year basically broke them completely. Theres a new method that doesn't use the -jni artifacts, but theres no documentation or templates on how to do them. I will try to make a template for the new way to build client apps this week. And we will update the documentation as well.

wrall, 2023-01-18:
> Theres a new method that doesn't use the -jni artifacts, but theres no documentation or templates on how to do them.

Do you have any hints while you write more formal documentation?  My team is hitting this too (sadly I couldn't find this issue before [posting to ChiefDelphi](https://www.chiefdelphi.com/t/issue-accessing-network-tables-on-linux-x64-coprocessor/423767)).

sciencewhiz, 2023-01-19:
One option would be to use the 2022 artifacts. The roborio's NT server will connect with NT 4 or NT 3 clients.

wrall, 2023-01-19:
Ah, fair enough.  I wasn't sure that compatibility was maintained between versions, glad to hear that it was.

lost1227, 2023-01-21:
We've been having the same problem running on our Raspberry Pi (`linuxarm64`).

I found a workaround using `LD_PRELOAD` to force `libwpiutil.so` to load:
```bash
wget https://frcmaven.wpi.edu/artifactory/release/edu/wpi/first/wpiutil/wpiutil-cpp/2023.2.1/wpiutil-cpp-2023.2.1-linuxarm64.zip
unzip -d wpiutil-cpp wpiutil-cpp-2023.2.1-linuxarm64.zip && rm wpiutil-cpp-2023.2.1-linuxarm64.zip
LD_PRELOAD=wpiutil-cpp/linux/arm64/shared/libwpiutiljni.so ./gradlew run
```

Note that this workaround is for `linuxarm64`. It should work the same on `linuxx86-64`, but you'll have to be sure to download the correct version of wpiutil.

rmheuer, 2023-01-23:
Building the project as in https://github.com/wpilibsuite/StandaloneAppSamples/blob/main/Java/build.gradle works properly for `linuxx86-64`.

closed 2023-01-23 by rmheuer