wpilibsuite/allwpilib #4984: AprilTagFieldLayout does not expect a asymmetric field

brennenputh, 2023-01-20:
The field tag layout is non-symmetric this year.  AprilTagFieldLayout makes assumptions it is.

Proposed solution: Add a transform to AprilTagFieldLayout that transforms correctly for the asymmetric tag layout

calcmogul, 2023-01-20:
No Transform3d you apply can reflect across an axis. AprilTagFieldLayout uses the North-West-Up coordinate system (x to the north, y to the west, z up). Mirroring across x or y isn't following the NWU coordinate system anymore.

The intended usage of this class is to choose one origin for everything (apriltags and pose estimation) and generate/transform your trajectories to match whatever your origin is.

Starlight220, 2023-02-07:
Are there any actionable items on this issue? (ie, can this be closed?)

closed 2023-08-03 by calcmogul