wpilibsuite/allwpilib #4994: Calculation Error of Rotation3d lead by floating point calculation error

richardthedou, 2023-01-21:
**Describe the bug**
After a Rotation3d has done several calculations, there could be an error in calculation (due to the floating point calculation mechanism in my opinion).

**To Reproduce**
Code to reproduce the behavior: 
`Rotation3d r1 = new Rotation3d(0,0,Math.PI/2);`
`Rotation3d r2 = new Rotation3d(Math.PI,0,0);`
`Rotation3d r3 = new Rotation3d(-Math.PI/2,0,0);`
`Rotation3d result = r1.plus(r2).plus(r3);`
`System.out.println(result);`
`System.out.println(result.getZ());`

**Expected behavior**
The true calculation result in Quaternion should be (0.5, 0.5,-0.5, 0.5). The print output is (0.5, 0.5000000000000001, -0.49999999999999994, 0.49999999999999994). When using `result.getZ()`, this gives an angle of about -0.24 radian which is about -14 degrees.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1
 - OS: Windows 10
 - Java version 17.0.5

**Additional context**
If any of the `Math.PI/2` is changed to something like `Math.PI/2-0.01`, then the result will be accurate.
[image]
[image]

calcmogul, 2023-01-21:
The conversion from quaternion to yaw Euler angle is:
```java
    // https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles#Quaternion_to_Euler_angles_(in_3-2-1_sequence)_conversion
    return Math.atan2(2.0 * (w * z + x * y), 1.0 - 2.0 * (y * y + z * z));
```

Those quaternion values result in `Math.atan2(-5.551115123125783e-17, 2.220446049250313e-16)`, and `Math.atan2()` has poor numerical conditioning when the arguments are near zero. We'll need to find a more stable conversion to fix it.

richardthedou, 2023-01-22:
Ohh interesting. If two inputs (x and y) of atan2 are zero, java gives the output of atan2(0,0) = 0, but is that mathematically correct? I'm not sure how Quaternions should be converted to Euler angles in this case.

richardthedou, 2023-01-24:
@PeterJohnson Yep that seems to work. I used a similar approach in my program and overridden the getX() and getZ() methods. Whenever the two parameters of atan2() are close enough to zero, I just reset them to zero.

PR #5437 (2023-07-18): [wpimath] Improve Euler angle calculations in gimbal lock
Fixes #4994

The `1e-20` threshold was just made up, it could be adjusted if desired.
closed 2023-07-18 by PeterJohnson