wpilibsuite/allwpilib #4997: Joystick buttons not registering in Simulator - Spektrum InterlinkDX Simulator Controller

team3572, 2023-01-22:
**Describe the bug**
Joystick buttons not registering in Simulator - Spektrum InterlinkDX Simulator Controller

**To Reproduce**
Steps to reproduce the behavior:
1. VC Code JAVA Code - Start Robot Simulator
2. Choose SIM GUI to start
3. InterlinkDX is in the list if available System Joysticks
4. Drag InterlinkDX to Joysticks Panel
5. Press buttons on joystick - no change in joystick panel

**Expected behavior**
As I click buttons I expect to see the joystick buttons change in the Joysticks panel.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.2.1
 - OS: Windows 11
 - Java version: What ever version is installed with the FRC Installation
 Also on Mac
OS: 12.6.2

All of the Axis of the joystick are updated in the Joystick panel
An XBox controller plugged in at the same time works as expected. All Axis are upadated and you can see all the button presses.
The robot code when run in simulation does not see the button presses either.

The driver station does see the joystick and all its buttons and the joystick works fine with code deployed to the RoboRio.

PeterJohnson, 2023-01-22:
Currently we rely on GLFW to access joysticks, so it's likely there's a compatibility issue there for Windows.  It's unclear to me whether the InterlinkDX has drivers for Mac, so it's probably something more fundamental there.

sciencewhiz, 2023-01-22:
It doesn't appear that anyone has contributed a mapping to https://github.com/gabomdq/SDL_GameControllerDB which is what I believe is used by GLFW.

team3572, 2023-01-22:
> Currently we rely on GLFW to access joysticks, so it's likely there's a compatibility issue there for Windows. It's unclear to me whether the InterlinkDX has drivers for Mac, so it's probably something more fundamental there.

Just to be clear this doesn't work on Windows or Mac. The behavior is the same. The Axis values are updated but not he button values.
But the buttons do work on RoboRio deployed robot code. The normal driver station sees the entire joystick perfectly fine.

team3572, 2023-01-22:
> It doesn't appear that anyone has contributed a mapping for that joystick to https://github.com/gabomdq/SDL_GameControllerDB which is what I believe is used by GLFW.

If a mapping is what might be necessary I'll go look at that repo and see if I can create one.

team3572, 2023-01-22:
So if I'm understanding this correctly an arbitrary controller has to be mapped to an Xbox style layout for it to work in the simulator?
I did a test with the GLFW itself and does read all the controller's buttons natively and in my robot program I'm accessing them using a CommandGenericHID.
The simulator does show all of the controller's 28 buttons in the GUI so it some level it is aware of all the joystick buttons it's just not showing the buttons states changing then the buttons are activated.

PeterJohnson, 2023-01-22:
No, it doesn't need to be mapped to an Xbox style layout.  If another GLFW application can see all the buttons, then you're right, it's something on our side, especially since the buttons show up but are just non-responsive.  What version of GLFW did you test with?

team3572, 2023-01-22:
I just used this utility someone had posted.
https://discourse.glfw.org/t/glfw-controller-database/768
It showed the controller fine - all the axis and button updated on their GUI in realtime.
Was made in 2016 so I'm guessing pretty old.

Was considering running the actual joystick test that is included in the GLFW repo today to see if there was any difference.
Would that help?

PeterJohnson, 2023-01-22:
I'm just not particularly finding where the issue could be in our code since the buttons count sounds like it's correct... that's being read by the same code that reads the buttons.

https://github.com/wpilibsuite/allwpilib/blob/main/simulation/halsim_gui/src/main/native/cpp/DriverStationGui.cpp#L375-L378

We do some munging of that later (in GlfwSystemJoystick::GetData()) but on Windows I'm not seeing anything that would cause the buttons not to be recognized.

team3572, 2023-01-24:
Would it be helpful to send you an actual controller?

PeterJohnson, 2023-01-24:
Yes, if you have one that you can loan, that would be the quickest solution.  You can private message me on Chief Delphi.

PR #5036 (2023-02-01): [simulation] GUI: Fix buffer overflow in joystick axes copy
This was using an incorrect sizeof which would copy excessive data and could overwrite the button data.

Fixes #4997.
closed 2023-02-01 by PeterJohnson