wpilibsuite/allwpilib #5005: DataLog example

rzblue, 2023-01-26:
Example to demonstrate the use of DataLog with both NT and `LogEntry`.

We could also use it for RLIs on the [frc-docs page](https://docs.wpilib.org/en/stable/docs/software/telemetry/datalog.html)

Starlight220, 2023-01-26:
I think it would be better to add this to a larger example than to create a whole new example just for logging. We should add a new tag for it, though.

PeterJohnson, 2023-01-26:
Per other discussions, I think we should have an area for documentation RLIs separate from the robot-level examples we include with vscode.  I think this would help make sure all the doc examples still compile without adding clutter to the vscode examples.

Starlight220, 2023-01-26:
That's a repo structure change that would need to wait for next year.

PeterJohnson, 2023-01-26:
We don't *have* to wait, as it would be non-breaking to teams.  But maybe it's too much RLI surgery / people are too busy right now?

rzblue, 2023-01-26:
I agree with all of the above, I just wanted to add an issue to track it because currently there are no examples that show off `DataLog`

Starlight220, 2023-01-26:
#5011 adds NT and DS logging to the Hatchbots, though it doesn't show use of the iterative `*Log` classes.