wpilibsuite/allwpilib #5009: Alternative Pose Estimators for Custom Drivetrains

egeakman, 2023-01-26:
As explained in https://docs.wpilib.org/en/stable/docs/software/advanced-controls/state-space/state-space-pose-estimators.html, WPILib includes the DifferentialDrivePoseEstimator, SwerveDrivePoseEstimator, and MecanumDrivePoseEstimator to estimate the robot's position. In our case, we have an unusual drivetrain, which we call H-Drive. It has four omni wheels, just like in a tank drive (two on each side), and two additional perpendicular omni wheels in the middle. This drivetrain style enables swerve-like movement.

[image]

However, the problem is that odometry is not supported by default. Is there any possible way or workaround for us to benefit from odometry? We know that we can use the DifferentialDrivePoseEstimator, but that means we cannot use the H-Drive motors for perpendicular movement during a season where perpendicular movement is a significant advantage.

Thanks in advance.

calcmogul, 2023-01-26:
The odometry and pose estimator classes use `Pose2d.exp(Twist2d)` to integrate encoder and gyro data. A Twist2d represents the chassis's x, y, and heading movement in the tangent space (the robot coordinate frame).

The Twist2d for a differential drive is:
```cpp
Twist2d twist{
  (leftEncoder + rightEncoder) / 2.0 - (prevLeftEncoder + prevRightEncoder) / 2.0,
  0.0,  // Differential drives can't move sideways in the robot coordinate frame
  heading - prevHeading};
```
The Twist2d for an H-drive is:
```cpp
Twist2d twist{
  (leftEncoder + rightEncoder) / 2.0 - (prevLeftEncoder + prevRightEncoder) / 2.0,
  centerEncoder - prevCenterEncoder,
  heading - prevHeading};
```

Autumn-Ou, 2023-04-01:
In regards to this post specifically a full HDrivePoseEstimator is available in Pull Request "HDrive Support! #5247" which includes odometry, pose estimation, kinematics, motor safety, prebuild drives, & simulation support.

egeakman, 2023-04-01:
> In regards to this post specifically a full HDrivePoseEstimator is available in Pull Request "HDrive Support! #5247" which includes odometry, pose estimation, kinematics, motor safety, prebuild drives, & simulation support.

Very cool! Our comp is finished but it would definitely help teams in the future. Will try to add C++ to the PR if I have the time.

Autumn-Ou, 2023-05-13:
> 

C++ part is done if you want to give it a test during the offseason, I'm currently just trying to get the linter to not yell at me for documentation.

Gold856, 2023-11-21:
OBE by 2024 drivetrain rules?

closed 2023-11-22 by Starlight220