wpilibsuite/allwpilib #5015: Downloaded WPILib today, getting error with Gradle Build

MukilSundaravadivel, 2023-01-27:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Download WPILib using instructions from documentation
2. After installation, clone existing repository and attempt to build

**Expected behavior**
Expected successful gradle build, successful installation of libraries

**Screenshots**
[image]

**Desktop**
 - WPILib Version: [2023.2.1]
 - OS: [Windows 11]
 - Java version [17.0.5]

**Additional context**
Add any other context about the problem here.

ThadHouse, 2023-01-27:
For your repo, you're not publishing the .gradle folder and the build folder, right? Having those folders stored in git is a very bad thing, and can break in many ways.

Just to double chck, can you create a new project, and does that work?

MukilSundaravadivel, 2023-01-28:
Just checked, cannot create a new project either. 
[image]
Thanks for the response!

ThadHouse, 2023-01-28:
Ok. Something is definitely wrong. Reboot your computer, go to your user directly and delete the .gradle folder. Then try things again.

MukilSundaravadivel, 2023-01-28:
> Ok. Something is definitely wrong. Reboot your computer, go to your user directly and delete the .gradle folder. Then try things again.

This fixed the problem, seems like it was just my laptop then. Thank you so much!

closed 2023-01-28 by sciencewhiz