wpilibsuite/allwpilib #5026: Ramsete examples should use SimpleMotorFeedforward.calculate(velocity, nextVelocity, dt)

calcmogul, 2023-01-29:
SimpleMotorFeedforward's `calculate(velocity, acceleration)` overload requires calculating acceleration manually (usually via backward finite difference). The overload that takes two velocities is more intuitive.

I think there's also instances of this in docs.wpilib.org's inline examples.

Starlight220, 2023-02-07:
How do we get the next velocity setpoint? I assume this is relevant for other uses of SimpleMotorFeedforward as well.

calcmogul, 2023-02-07:
> How do we get the next velocity setpoint?

A motion profile. You'd call Trajectory.Sample() with a time that's one timestep in the future.

Starlight220, 2023-02-07:
Current API shape of anything doesn't expose that.

calcmogul, 2023-02-07:
Trajectory.sample() does.

Starlight220, 2023-02-07:
Wouldn't sampling a future state have problems at the last timestamp?

I'm talking about RamseteCommand, TrapezoidProfileCommand/Subsystem, ProfiledPID, etc.

calcmogul, 2023-02-07:
> Wouldn't sampling a future state have problems at the last timestamp?

No.
```java
  public State sample(double timeSeconds) {
    if (timeSeconds <= m_states.get(0).timeSeconds) {
      return m_states.get(0);
    }
    if (timeSeconds >= m_totalTimeSeconds) {
      return m_states.get(m_states.size() - 1);
    }
```

Gold856, 2024-06-18:
OBE by #6494.

closed 2024-06-18 by calcmogul