wpilibsuite/allwpilib #5027: LTVDifferentialDriveController constructor never completes in Java

DeltaDizzy, 2023-01-29:
**Describe the bug**
When using the Java version LTVDifferentialDriveController, the constructor takes a very long time to complete. On our RIO 1 it ran for a little over 6 and a half minutes before the JVM crashes with an out of memory error. In sim, it ran for over 11 and a half minutes before i stopped it, still without finishing the constructor.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new WPILib Java TimedRobot project with desktop support
2. Edit Robot.java to contain a single instantiation of an LTVDifferentialDriveController and print statements before and after it (such as [here](https://gist.github.com/DeltaDizzy/1191cc0bdcf0206b69161869053415bc)). The LinearSystem constants and trackwidth used were determined via SysId (albeit a java logger impl) on a real drivetrain.
3. Simulate the code and note that the first print statement prints but the second one does not.

**Expected behavior**
The two print statements should both print within a few seconds of each other, indicating that the LTVDifferentialDriveController object was successfully constructed and is ready for use.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.2.1
 - OS: Windows 10
 - Java version: 17.0.5 (but with 11 compatibility set in gradle)

calcmogul, 2023-01-29:
A linear Kv of 0.00052047 m/s means that 12V results in a steady-state velocity of 23,056 m/s. The LUT is generated in 0.01 m/s increments, so there's 2.3056 million entries. On my machine, the LTV LQR instance takes 30 ms to construct each, so that's a total runtime of 19 hours.

calcmogul, 2023-01-29:
We should also look into smarter interpolation schemes so we don't need so many LUT entries.

commit 21439b60 (2023-08-03): [wpimath] Disallow LTV controller max velocities above 15 m/s (#5495)

15 m/s is about 50 ft/s, which is way above what FRC robots should be
able to achieve. This limit lets us catch user errors from bad unit
conversions immediately instead of the LUT generation in the LTV
controllers hanging for a really long time.

Fixes #5027.
closed 2023-08-03 by PeterJohnson