wpilibsuite/allwpilib #5034: Wrong axis in XboxController class

AndromedaHelix, 2023-01-31:
**Describe the bug**
Today I was working on the chassis code, when I found a persistent issue while using `XboxController.getRightY` & `XboxController.getRightX`. These values were being registered even when nothing was pressed. After some thorough debugging, I found that the axis ports used for these was used wrongly and changed with those of the bumpers. Here are photos that further explain this:

**To Reproduce**
Steps to reproduce the behavior:
1. Just create a `XboxController` object and use the `getRightX` and `getRightY` functions. 

**Expected behavior**
I would expect for the values to represent what the current joystick's value is. 

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[image]
[image]

**Desktop (please complete the following information):**
 - WPILib Version: 2021.3.1
 - OS: MacOS
 - Java version 1.10.2

ThadHouse, 2023-01-31:
You need to click the Map Gamepad checkbox on macOS and linux sometimes to fix the axes. Different operating systems have different mappings for joysticks by default.

sciencewhiz, 2023-01-31:
Why are you using WPILib from 2021?

AndromedaHelix, 2023-01-31:
Sorry, that was a typo. I am using wpilib 2023

Starlight220, 2023-02-13:
Can this be closed?

AndromedaHelix, 2023-02-13:
Yes, just tried it out. It appears to be an error with the simGUI

closed 2023-02-13 by AndromedaHelix