wpilibsuite/allwpilib #5035: Simulation Motors

NetLockJ, 2023-01-31:
**Describe the bug**
New motors seem not to appear in the simulation list of "Other devices" in the Simulation GUI. Even if ports of existing motors are changed, they don't change on the Simulation GUI. I've tried restarting VS Code, but it  doesn't seem to be fixing the issue.

**To Reproduce**
Steps to reproduce the behavior:
- Make a motor that hasn't been created and simulated on in code before (or change a port of an existing one)
- Run simulation

**Expected behavior**
New motors should appear, and ones with edited ports should change

**Screenshots**

Motors added in port 1 and 2

[image]

Motors don't show up on the list..

[image]

Changing the port of one of the motors that *do* exist will also yield the same results. So, if I changed 10 to 11 for example, it will still show up on the simulation as 10.

**Desktop (please complete the following information)**
 - WPILib Version: [Latest version]
 - OS: [Mac OS Montery 12.6]
 - Java version [openjdk 17.0.5 2022-10-18]

auscompgeek, 2023-01-31:
CTRE Phoenix devices only register themselves with WPILib's simulation support for the `WPI_*` classes, i.e. `WPI_TalonSRX`, `WPI_TalonFX`, etc.

NetLockJ, 2023-01-31:
Oh, that makes so much more sense! Thanks!

closed 2023-01-31 by Starlight220