wpilibsuite/allwpilib #5037: Network Tables

brhea, 2023-01-31:
Our code uses networktable entries and I've noticed that persistent values do not seem to be persistent after reboot. I found the following statement in the documentation for network table programming: 

"Some functionality (e.g. persistent settings) has also moved to Topic properties (see [NetworkTables Tables and Topics]"
(https://docs.wpilib.org/en/latest/docs/software/networktables/tables-and-topics.html#networktables-tables-and-topics)).

However, setPersistent() is still a method of network table entries and is in the javadoc. 

Do I have to rewrite using Topics to get persistence to work?

PeterJohnson, 2023-01-31:
No, you shouldn’t need to.  Can you share your snippet of code that’s setting up persistent entries?

brhea, 2023-01-31:
There's a fair amount of code around this but basically it's just:

            if(shouldKeepOnReboot) ntEntry.setPersistent();  // make value persist through reboots
            else ntEntry.clearPersistent();

This code has run for a few years now. Please don't spend any time on this until I've further investigated. We're really not spent any time debugging this due to other priorities. I was just reading through the documentation to try to understand the changes and see any clues to our issues and noted the phrase that "functionality moved" and thought it may have been removed from the networktable entries.

This could be a timing thing also. We probably have a fair number of persistent entries. Does it take a while for all to write to a file if there are a lot of them. Many of these are PID tuning parameters so they don't change frequently.

PeterJohnson, 2023-02-07:
I added a warning in the latest release for this case.  Essentially the problem is in NT4 it’s necessary to do a setDefault() (or set()) before calling setPersistent().  There’s not a good way to restore the original NT3 behavior where this wasn’t required.

PeterJohnson, 2023-02-08:
Yes.

Starlight220, 2023-02-14:
Is this resolved?

brhea, 2023-02-14:
Yes

closed 2023-02-14 by brhea