wpilibsuite/allwpilib #5039: Refactor HID classes for expandability and move to their own subdirectory

Oblarg, 2023-02-02:
While our HID classes are fundamentally very similar, there is a lot of code duplication between them; essentially all that varies between different HID classes is a simple set of button-mapping-related metadata, but we manually write analogous API shapes for each type of supported controller from this metadata rather than defining a suitably-generic abstraction that allows the class to be defined from such a minimal description in code.

This greatly limits our ability to expand our range of supported HID interfaces, despite the fact that additional HID descriptions *should* be very simple (again, we're only really adding a little bit of mapping metadata).

Additionally, our HID classes currently live in base-level WPILibJ/C.  This is also a limitation to expansion/generalization, and HID-related code should probably all be moved to a common package/subdirectory.

ncgears, 2023-02-02:
Details of Stadia Controller mapping, for when this moves forward:

```
        /**
         * This class defines the hardware button and axis IDs for a Stadia Controller.
         * The buttons array is 1-based, but the axis array is 0-based
         */
        public static final class Stadia {
            //DO NOT EDIT THESE
            static final int BTN_A = 1; //A Button
            static final int BTN_B = 2; //B Button
            static final int BTN_X = 3; //X Button
            static final int BTN_Y = 4; //Y Button
            static final int BTN_LB = 5; //Left Bumper (L1)
            static final int BTN_RB = 6; //Right Bumper (R1)
            static final int BTN_L = 7; //Left Stick Press (L3)
            static final int BTN_R = 8; //Right Stick Press (R3)
            static final int BTN_ELLIPSES = 9; //Ellipsis Button (...)
            static final int BTN_HAMBURGER = 10; //Hamburger Button
            static final int BTN_STADIA = 11; //Stadia Button
            static final int BTN_RT = 12; //Right Trigger (R2)
            static final int BTN_LT = 13; //Left Trigger (L2)
            static final int BTN_GOOGLE = 14; //Google Button
            static final int BTN_FRAME = 15; //Square Frame Button
            static final int AXIS_LH = 0; //Left Analog Stick horizontal (right +)
            static final int AXIS_LV = 1; //Left Analog Stick vertical (down +)
            static final int AXIS_LV2 = 2; //Left Analog Stick vertical (down +) - duplicate
            static final int AXIS_RH = 3; //Right Analog Stick horizontal (right +)
            static final int AXIS_RV = 4; //Right Analog Stick vertical (down +)
            static final int AXIS_RV2 = 5; //Right Analog Stick vertical (down +) - duplicate
            static final int DPAD_UP = 0;
            static final int DPAD_UPRIGHT = 45;
            static final int DPAD_RIGHT = 90;
            static final int DPAD_DNRIGHT = 135;
            static final int DPAD_DN = 180;
            static final int DPAD_DNLEFT = 225;
            static final int DPAD_LEFT = 270;
            static final int DPAD_UPLEFT = 315;
            static final int DPAD_IDLE = -1; 
        }

```

ncgears, 2023-02-02:
Details of Logitech F310 Controller mapping
```

        /**
         * This class defines the hardware button and axis IDs for a Logitech F310 Controller.
         * The buttons array is 1-based, but the axis array is 0-based
         */
        public static final class Logitech {
            //DO NOT EDIT THESE
            static final int BTN_A = 1; //A Button
            static final int BTN_B = 2; //B Button
            static final int BTN_X = 3; //X Button
            static final int BTN_Y = 4; //Y Button
            static final int BTN_LB = 5; //Left Bumper (L1)
            static final int BTN_RB = 6; //Right Bumper (R1)
            static final int BTN_BACK = 7; //Back Button (Select)
            static final int BTN_START = 8; //Start Button
            static final int BTN_L = 9; //Left Stick Press (L3)
            static final int BTN_R = 10; //Right Stick Press (R3)
            static final int AXIS_LH = 0; //Left Analog Stick horizontal
            static final int AXIS_LV = 1; //Left Analog Stick vertical
            static final int AXIS_LT = 2; //Analog Left Trigger
            static final int AXIS_RT = 3; //Analog Right Trigger
            static final int AXIS_RH = 4; //Right Analog Stick horizontal
            static final int AXIS_RV = 5; //Right Analog Stick vertical
            static final int DPAD_UP = 0;
            static final int DPAD_UPRIGHT = 45;
            static final int DPAD_RIGHT = 90;
            static final int DPAD_DNRIGHT = 135;
            static final int DPAD_DN = 180;
            static final int DPAD_DNLEFT = 225;
            static final int DPAD_LEFT = 270;
            static final int DPAD_UPLEFT = 315;
            static final int DPAD_IDLE = -1; 
        }
```

calcmogul, 2023-02-02:
So you're suggesting code generation from e.g., JSON, YAML, TOML descriptions?

Oblarg, 2023-02-02:
yeah, more or less

Starlight220, 2023-02-02:
In C++, macros make this less of a problem. Java is the real pain (and it's the larger userbase).

As for a `.hid` package, we've discussed this last year and the idea was shot down due to the HID acronym not being known enough.

calcmogul, 2023-02-02:
> In C++, macros make this less of a problem

But you'd need to inspect what the macro does to find out how to call the functions it creates. We can't generate API documentation if the class is generated at compile time.

> As for the Logitech, I was under the impression that it's mapping was identical to an XboxController.

It was either that or the PS4Controller. There's a "D/X" switch on the back to switch between DirectInput and XInput mappings.

sciencewhiz, 2023-02-03:
Could potentially generate from https://github.com/gabomdq/SDL_GameControllerDB, if we want 1000 HID classes

Starlight220, 2023-06-25:
#5257 is proving exactly why this is needed. Do we intend on moving this forward?

Gold856, 2024-06-17:
Closed by #6274

closed 2024-06-17 by calcmogul

auscompgeek, 2024-06-17:
This issue was suggesting that HIDs should all move to a subpackage too, right? I don't think that's been done. I think having all of them in an `frc::hid` namespace / `wpilibj.hid` package would be good for discoverability.

KangarooKoala, 2024-06-17:
> This issue was suggesting that HIDs should all move to a subpackage too, right?

[This earlier comment](https://github.com/wpilibsuite/allwpilib/issues/5039#issuecomment-1414204770) indicated why the hid package was not considered:

> As for a `.hid` package, we've discussed this last year and the idea was shot down due to the HID acronym not being known enough.