wpilibsuite/allwpilib #5066: Motors will disable-blip on first enable in certain circumstances

CoryNessCTR, 2023-02-06:
**Describe the bug**
The first time you DS-Enable (after power-cycle or code deploy), the motors drive for a split second before disabling for about a second, then resuming.
Subsequent enables do not reproduce this.

**To Reproduce**
Steps to reproduce the behavior:
1. Copy below code to new Timed Skeleton (Advanced) project
2. Deploy project
3. Enable teleop (motor should drive for a split-second, disable for about a second, then continue driving)
4. Subsequent disable-enables do not reproduce
5. Redeploy or restart RIO and repeat step 3

```java
/* Start of new Timed Skeleton (Advanced) code */
public class Robot extends TimedRobot {
  PWMTalonFX _motor1 = new PWMTalonFX(0);
  PWMTalonFX _motor2 = new PWMTalonFX(1);

  DifferentialDrive _diff = new DifferentialDrive(_motor1, _motor2);

  String work = ""; // Could be declared here or in teleopPeriodic
  @Override
  public void teleopPeriodic() {
    double forward = 0.5; // Double assignment necessary to reproduce (Can't just use literal 0.5 everywhere)
    _diff.arcadeDrive(forward, 0); // Differential drive set necessary to reproduce (Can't use _motor1 directly)
    work = " forward value: " + forward; // String concatenation necessary to reproduce (Comment this out and it does not reproduce)
  }
```

**Expected behavior**
Motors should not disable-blip

**Desktop (please complete the following information):**
 - WPILib Version: 2023.3.2
 - OS: Windows 10

**Additional context**
Issue originally reported in CD thread https://www.chiefdelphi.com/t/canivore-stops-communicating-with-devices-after-first-enable/425891

calcmogul, 2023-02-06:
I've seen this behavior with Spark Maxes too since at least 2020. I can't remember past that.

rzblue, 2023-02-06:
It looks like this is caused by slow initial string concatenation in java, combined with MotorSafety being enabled by default on the drive classes. The string concatenation blocks the thread on the first invocation, which causes MotorSafety to time out and disable the motors.

Gold856, 2026-01-16:
I wonder if Systemcore is fast enough for this to no longer be an issue.