wpilibsuite/allwpilib #5069: NetworkTable value issues between client and server

webbbt456, 2023-02-07:
I'm attempting to create a NetworkTable client user interface (Windows, MFC) for our dashboard. The UI code worked using last years NetworkTable code.  I'm having troubles with values propagating between the server and client. 

Attaching code dealing with connecting, multisubscriber, poller, and connection events.

On the client side, how do I detect that the connection to the server has been lost? I never get a connection event on the client.
What is the best method to reconnect to the server without having to restart the dashboard client? The previous version of NetworkTables would automatically reconnect.

How do I detect that the multisubscriber on the tables have worked? I never get an event from the poller ReadQueue call. 

Before I added the multisubscriber and poller, the existing values on the server would not propagate from the server to the client.  The first value being "read" from the Preferences table would always be reported as not contained or exiting in the table. After I added the multisubscriber and poller,  the first value being "read" from the Preferences table may be reported as not contained or exiting in the table.

Any information would be greatly appreciated, Thanks

[NetworkTableCode.txt](https://github.com/wpilibsuite/allwpilib/files/10670864/NetworkTableCode.txt)

PeterJohnson, 2023-02-07:
Connection events are only issued on the instance handle, not on other handles.  Are you using 2023.3.2 dependencies?

webbbt456, 2023-02-07:
Using the latest master branch from last night after getting updated release.

I have it working now.  It works without the leading /, changed it to make sure.  Started getting server disconnects and it reconnects automatically.  Removed the thread that was calling ReadQueue.  I also changed the code from getting the value of the boolean /SmartDashboard/clientName and used a BooleanSubscriber to get the value that will be set by the server (i.e. robot) when it gets a connection event.

Attaching the new code
Thanks for the help.
[NetworkTableCode2.txt](https://github.com/wpilibsuite/allwpilib/files/10671600/NetworkTableCode2.txt)

closed 2023-02-08 by webbbt456