wpilibsuite/allwpilib #5071: SendableChooser doesn't work with move-only types such as CommandPtr

Starlight220, 2023-02-07:
This has multiple parts, not all of them necessarily in this repo.

In this repo: if there's a way to improve the compile error to something more understandable, that would be a great improvement. (maybe something with `static_assert`?). At the minimum we should document that it doesn't work with move-only types.

As for the very common use case of using it for auto command selection, here are the options:
- use `SendableChooser<Command*>`, and store all the relevant CommandPtrs in RobotContainer, passing them as options using `CommandPtr.get()` (it returns a `Command*`). The HatchbotInlined example can be used as a reference on how to do this.

- Use `SendableChooser<CommandPtr()>`; this will select between factories, and then you call the chosen factory at autoInit and schedule the constructed command. This is likely the preferred option (for Java as well), as it makes it harder to accidentally re-compose the same instances.

- Use `SelectCommand<T>` with a `SendableChooser<T>` as the selector function: `T` (used as the `SelectCommand` key type and the `SendableChooser` value type) can be any type you want. One possible pitfall here is that the options need to be added to both the SelectCommand and SendableChooser, using an enum as `T` might be able to help prevent mismatches.

We should document at least one of these.

nuttle, 2023-02-09:
> Use `SendableChooser<CommandPtr()>`

This seems like the cleanest option -- however, I think there may be a problem here.  I could certainly be wrong -- I have not had time to dig into this.

In `frc/smartdashboard/SendableChooser.h`:

```
template <class T>
class SendableChooser : public SendableChooserBase {
  wpi::StringMap<T> m_choices;

  template <class U>
  static U _unwrap_smart_ptr(const U& value);

  template <class U>
  static U* _unwrap_smart_ptr(const std::unique_ptr<U>& value);

  template <class U>
  static std::weak_ptr<U> _unwrap_smart_ptr(const std::shared_ptr<U>& value);

.
.
.
```

Not having something that matches `CommandPtr` may be leading to problems.  In particular, Just declaring `frc::SendableChooser<frc2::CommandPtr()> xyz;` seems to cause compilation problems.  Is there an example of this usage somewhere?  Does this seem to be off somehow?

Here's a simple reproduction case (just create a minimal new C++ project and add these lines):

```
#include <frc/smartdashboard/SendableChooser.h>
#include <frc2/command/CommandPtr.h>

void Robot::RobotInit() { frc::SendableChooser<frc2::CommandPtr()> xyz; }
```

Starlight220, 2023-02-09:
Does `SendableChooser<std::function<CommandPtr>>` work?

PeterJohnson, 2023-02-09:
That's missing a std::function, it needs to be `frc::SendableChooser<std::function<frc2::CommandPtr()>>`

nuttle, 2023-02-09:
Yes, thank you both -- that works.  I actually tried this before, but must have had something else going on in the non-trivial test case.  I should be able to get things working from here.  Thanks again!

Starlight220, 2023-02-09:
I've now looked at the source code, and it seems that LLVM's `StringMap` is the one imposing the restriction that `T` must be copy- and default-constructible.

calcmogul, 2023-02-09:
If std::map meets our requirements better, we could use that instead.

Starlight220, 2023-02-09:
Wouldn't any collection for this cause a problem with move-only types? Since we don't want to move the object out of the collection.

calcmogul, 2023-02-09:
I was responding to this statement in particular:
> I've now looked at the source code, and it seems that LLVM's StringMap is the one imposing the restriction that T must be copy- and default-constructible.

If StringMap not supporting move-only types is a problem, we can fix that by using an STL container that does support move-only types.

Starlight220, 2023-02-09:
I can look into it, though I think that choosing between suppliers is better anyway: we've seen multiple recomposition errors due to people composing the selected command.

calcmogul, 2023-02-09:
True, returning an existing command instance is error-prone.

PR #5078 (2023-02-12): [wpilibc] SendableChooser: static_assert copy- and default-constructibility
Resolves #5071 with a more obvious error message.
closed 2023-02-12 by PeterJohnson