wpilibsuite/allwpilib #5074: NetworkTable GetTopicInfo, GetTopics, GetKeys, and GetSubTables do not return the expected values

webbbt456, 2023-02-08:
I am attempting to implement NetworkTable Preferences Load and Save to file. I am also trying to figure a way to remove all of the preferences stored on the robot.  I am attempting to call frc::Preferences::RemoveAll, the call does not remove anything. Running the code in VS2022 debugger I stepped into the call finally getting to the NetworkTable::GetKeys method.  It only returns the ".type" entry in the returned value.  I then added code to directly call the GetKeys and GetSubTables method using std::shared_ptr<nt::NetworkTable> of the Preferences table. I get the same results for the GetKeys method.  The GetSubTables returns a list equal to the number of entries in the Preferences table containing the same value (only one SubTable).  I was expecting to get a list of unique SubTables of the Preference Table. I have not tested the GetTopics or GetTopicInfo methods, they are identical to the GetKeys method except for the return value.  

I have modified the methods (provided below), the modifications give me the results that I expect.  frc::Preferences::RemoveAll still does not remove any of the entries in the Preferences table.

```
std::vector<TopicInfo> NetworkTable::GetTopicInfo(int types) const {
  std::vector<TopicInfo> infos;
  size_t prefix_len = m_path.size() + 1;
  for (auto&& info :
       ::nt::GetTopicInfo(m_inst, fmt::format("{}/", m_path), types)) {
    auto relative_key = wpi::substr(info.name, prefix_len);
    if (relative_key.ends_with(PATH_SEPARATOR_CHAR)) {
      continue;
    }
    infos.emplace_back(std::move(info));
  }
  return infos;
}

std::vector<Topic> NetworkTable::GetTopics(int types) const {
  std::vector<Topic> topics;
  size_t prefix_len = m_path.size() + 1;
  for (auto&& info :::nt::GetTopicInfo(m_inst, fmt::format("{}/", m_path), types)) {
    auto relative_key = wpi::substr(info.name, prefix_len);
    if (relative_key.ends_with(PATH_SEPARATOR_CHAR)) {
      continue;
    }
   topics.emplace_back(info.topic);
  }
  return topics;
}

std::vector<std::string> NetworkTable::GetKeys(int types) const {
  std::vector<std::string> keys;
  size_t prefix_len = m_path.size() + 1;
  for (auto&& info : ::nt::GetTopicInfo(m_inst, fmt::format("{}/", m_path), types)) {
    auto relative_key = wpi::substr(info.name, prefix_len);
    if (relative_key.ends_with(PATH_SEPARATOR_CHAR)) {
      continue;
    }
   keys.emplace_back(relative_key);
  }
  return keys;
}

std::vector<std::string> NetworkTable::GetSubTables() const {
  std::string lastSubTable(""), subTable;
  std::vector<std::string> keys;
  size_t prefix_len = m_path.size() + 1;
  for (auto&& topic : ::nt::GetTopicInfo(m_inst, fmt::format("{}/", m_path), 0)) {
    auto relative_key = wpi::substr(topic.name, prefix_len);
    size_t end_subtable = relative_key.find(PATH_SEPARATOR_CHAR);
    if (end_subtable == std::string_view::npos) {
      continue;
    }
    subTable = wpi::substr(relative_key, 0, end_subtable);
    if (lastSubTable != subTable) {
      keys.emplace_back(subTable);
      lastSubTable = subTable;
    }
  }
  return keys;
}

```
**Desktop** 
 - WPILib Version:  2023.3.2
 - OS: Windows 10
 - C++ version 20

**Additional context**
Add any other context about the problem here.

PeterJohnson, 2023-02-08:
Agreed that GetSubTables() needs fixing to return effectively a set with no duplicates (this is what the Java version does), but I think the other functions are correct as written--the intent of them is to only return the keys in the immediate table, not any keys that are in subtables.  What are the keys of your preferences?  Are they all in subtables?

Arguably RemoveAll *should* go through all the subtables as well, but it's always behaved this way (only going through the local table).

In NT4, Remove/RemoveAll just sets the persistent value to false and unpublishes locally.  If something else on the network is also publishing, the key won't go away immediately--but it will no longer be persistent.  There's no way in NT4 to actually delete anything, as you can't force another publisher to stop publishing.

webbbt456, 2023-02-08:
All of my Preferences keys are in subtables.  I'm not trying to remove keys during normal operations. I want to clear the Preferences and load a set of stored values from a json file. Is there any way to deploy the json file to the roborio to replace the current file being used. I'm really looking for a way, other than reimaging the roborio, to start with an empty NetworkTables. Could I stop and start the server on the robot after removing all of the Preferences. I see the StartServer method allows for a json file to be specified, Is it possible to have it load from the specified file and then use the default file for normal operations. Most of this deals with transferring the NetworkTable contents to a new sd card with updated code. In the past I used the Load and Save methods for frc::Preferences.

Would it be useful to add NetworkTable::GetAllKeys, NetworkTable::GetAllTopics, and NetworkTable:: GetAllTopicInfo? The GetAll methods would allow for a non-recursive traversal of a Table/SubTable. It might also be useful to have a NetworkTable::GetAllSubTables that returns the list of all descendant SubTables of the NetworkTable . I could also see a use for a NetworkTable::AddAllListener that calls the callback on any key that is a descendant of the NetworkTable.

PeterJohnson, 2023-02-09:
There is no current method to do that, but it's on the wishlist (#4719).  You also can't specify a different file for auto-save than loading.

You can deploy a new .json file with SSH/SFTP (login lvuser, no password, location of the file is in /home/lvuser), but you're right you'll want to kill the robot program while doing that.  Killing the robot program can be done via ssh as well, I think it's `/usr/local/frc/frcKillRobot.sh -t` and then the same command with `-r` to restart the robot.

Yes, I can see the utility of adding those functions.