wpilibsuite/allwpilib #5080: NetworkTables 4 Subscription not sent all values

tsar-boomba, 2023-02-11:
**Bug Description**
I make a NT4 subscription with an empty string `""`, prefix, and all, but the server doesn't send my subscription all the values currently recognized. All topics do get announced to the client, but only about 19 - 24 out of 30 actually have their current value sent to my client.

**To Reproduce**
Steps to reproduce the behavior:
1. Subscribe with `[""]` and `{"prefix": true, "all": true}`
2. Not all values in the server are sent to the client

**Expected behavior**
All currently recognized topic values are to be sent to a client that subscribes to all values when they subscribe.

**Desktop:**
 - WPILib Version: 2023.2.1
 - OS: MacOS 12.3
 - Java version: 11

**Additional context**
I am using my own implementation of the NT4 spec in [this repo](https://github.com/tsar-boomba/network-tables-rs), although this shouldn't be affecting anything because everything else works as expected.

PeterJohnson, 2023-02-11:
Upgrade to 2023.3.2.  This should already be fixed there.

tsar-boomba, 2023-02-11:
I updated to 2023.3.2, restarted my laptop, and updated my robot project to use 2023.3.2. However, I still have the same issue as before. Below is a screenshot of the problem and the subscription I'm using.

[image]
[image]

PeterJohnson, 2023-02-12:
Do you have any other subscriptions already created before this subscription?

tsar-boomba, 2023-02-12:
This is the only subscription. The topics that I don't receive do have values, I've been working around this by restarting my client until I get the values I need.

PeterJohnson, 2023-02-13:
Do you have a link to your robot project?  I’m puzzled as to how this can be happening looking at the code that handles sending initial announcements and values.

tsar-boomba, 2023-02-13:
This is the repo and branch I'm using https://github.com/Yeti-Robotics/aurora-java-2022/tree/development

sciencewhiz, 2023-02-13:
That branch is still using 2023.1.1

tsar-boomba, 2023-02-13:
Sorry, that is because I haven't pushed the change to 2023.3.2 in the `build.gradle`. On my machine, it is running 2023.3.3.2

PeterJohnson, 2023-02-14:
So testing your robot program in simulation, there are no values being published to several published topics:
- /Shuffleboard/.metadata/Selected
- /Shuffleboard/.recording/FileNameFormat
- /Shuffleboard/.recording/RecordData
- /CameraPublisher/USB Camera 0/modes

The sim GUI NT view shows they don't have values, and no value is sent over the network.

I ran the C++ client with some WS debug-level logging turned on: https://gist.github.com/PeterJohnson/9796cab6a78438680383ebf96117fccc

I'm not sure why you're seeing a discrepancy of 16.  I count 33 announced and 29 values sent in the above.

tsar-boomba, 2023-02-14:
Alright, I'll look over my implementation, I must have missed something there. Thank you for your help!

closed 2023-02-14 by tsar-boomba

tsar-boomba, 2023-02-14:
Yeah, looks like a lot of values are being received before the server announces them, so I've gotta solve that now, thank you!

PeterJohnson, 2023-02-14:
Huh?  Values aren’t sent before they are announced (see the trace above, each announcement is followed by the value).

tsar-boomba, 2023-02-14:
Yes, they are received in order, but I process messages asynchronously which is what caused the discrepancy. I'm definitely considering changing this behavior in the future, but for now, I have it worked out. Once again, thank you for your help.