wpilibsuite/allwpilib #5088: Filter design tool

calcmogul, 2023-02-12:
**Is your feature request related to a problem? Please describe.**
It's difficult for teams to select and tune filters to meet their signal processing requirements.

**Describe the solution you'd like**
I'd like to a see a filter design tool akin to the one in MATLAB's DSP toolbox that supports the following:

* Read in sensor data from pre-recorded .wpilog files
* Read in sensor data live over NetworkTables via NT4 topics
* Export to LinearFilter class coefficients
* Common filters like Butterworth, Chebyshev, Elliptical, Notch (could be part
  of the LinearFilter factory functions)
* Building higher order filters from biquad segments
* FFT and/or Morlet wavelets of data to help user determine cutoff frequencies
* Pole placement?
* Show pole-zero map?

**Describe alternatives you've considered**
A filter base class was suggested in https://github.com/wpilibsuite/allwpilib/issues/5083, but that still requires manual user plumbing to dashboard elements that could be done in a dedicated filter UI instead.

Starlight220, 2023-02-12:
Could this be made part of SysId?

calcmogul, 2023-02-12:
SysId is for a completely different use case.