wpilibsuite/allwpilib #5098: Swerve examples should use absolute encoders for azimuth control

Starlight220, 2023-02-14:
Currently they use quadrature encoders.
The examples should be changed to use `DutyCycleEncoder`/`AnalogEncoder`, and maybe even use Preferences to store azimuth offsets.

sciencewhiz, 2023-12-28:
Came up here also: https://www.chiefdelphi.com/t/multiple-ports-channels-for-encoder-corresponding-wiring/446960/3?u=joe_ross

sciencewhiz, 2024-02-03:
https://www.chiefdelphi.com/t/has-anyone-swervedrive-in-robotpy-using-neos-and-ctre-encoders/452715/2

spacey-sooty, 2024-06-23:
Don't DutyCycleEncoder and AnalogEncoder provide no equivalent to GetRate?
Should that be added?

ThadHouse, 2024-06-23:
> Don't DutyCycleEncoder and AnalogEncoder provide no equivalent to GetRate? Should that be added?

It’s not possible to add a get rate to dce and ae. The way the measure isn’t timed based so getting a rate is completely manual.

spacey-sooty, 2024-06-23:
> > Don't DutyCycleEncoder and AnalogEncoder provide no equivalent to GetRate? Should that be added?
> 
> It’s not possible to add a get rate to dce and ae. The way the measure isn’t timed based so getting a rate is completely manual.

Should the examples then show getting the rate from it?