wpilibsuite/allwpilib #5114: WPILIB 2023.4.1 RobotBase does not implement MathShared.getTimeStamp() for MathSharedStore

Zoarial94, 2023-02-18:
**Describe the bug**
Running `gradlew :simulateJava` opens the simulation GUI briefly, but the exception follow exception is thrown and causes the window to close: 

`Unhandled exception: java.lang.AbstractMethodError: Receiver class edu.wpi.first.wpilibj.RobotBase$2 does not define or inherit an implementation of the resolved method 'abstract double getTimestamp()' of interface edu.wpi.first.math.MathShared.`

Reverting back to 2023.3.2 allows the simulation to continue to run.

**To Reproduce**
Steps to reproduce the behavior:
1. Update the gradle WPILIB plugin to "2023.4.1"
2. Run ./gradlew :simulateJava

**Expected behavior**
Simulator GUI opens and does not crash

**Desktop (please complete the following information):**
 - Personal Machine
   - WPILib Version: 2021.4.1
   - OS: Linux Fedora 37
   - Java version (WPILIB 2023 provided JDK)
     - IMPLEMENTOR="Eclipse Adoptium"
     - IMPLEMENTOR_VERSION="Temurin-17.0.5+8"
     - FULL_VERSION="17.0.5+8"
     - SEMANTIC_VERSION="17.0.5+8"
     - BUILD_INFO="OS: Linux Version: 5.8.0-43-generic"
     - JVM_VARIANT="Hotspot"
     - JVM_VERSION="17.0.5+8"
 - Student's Windows Machine
   - WPILib Version: 2021.4.1
   - OS: Windows 10
   - Java version (WPILIB 2023 provided JDK)

**Additional context**
Our team uses Intellij and the FRC plugin offered to update the WPILIB library in gradle. We will be reverting back to 2023.3.2 until 2023.4.1 has been officially released. If you are not taking issues for 2023.4.1, feel free to close this issue.

PeterJohnson, 2023-02-18:
Except it does… https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/RobotBase.java#L134

PeterJohnson, 2023-02-18:
Maybe try a clean and build again?

Zoarial94, 2023-02-18:
You are correct, this is my mistake. We are using [AdvantageKit](https://github.com/Mechanical-Advantage/AdvantageKit) which reimplements certain base classes. I got ahead of myself here. Thank you for the quick response.

closed 2023-02-18 by Zoarial94

woofer300, 2023-02-18:
> You are correct, this is my mistake. We are using [AdvantageKit](https://github.com/Mechanical-Advantage/AdvantageKit) which reimplements certain base classes. I got ahead of myself here. Thank you for the quick response.

How did you resolve this issue? My team is having the same problem.

Zoarial94, 2023-02-18:
@woofer300 if you're using VSCode, there is an option to update the vendordep. Otherwise, you need to update the 4 versions in vendordeps/AdvantageKit.json to 2.2.0.