wpilibsuite/allwpilib #5120: Bug: WPILib logger crashes with segmentation fault when running Google Tests

gaborszita, 2023-02-20:
**Describe the bug**
WPILib logger crashes with segmentation fault when running Google Tests.

I used the [Elevator Simulation example](https://github.com/wpilibsuite/allwpilib/tree/main/wpilibcExamples/src/main/cpp/examples/ElevatorSimulation) on the WPILib C++ examples and modified it so that it uses the [WPILib logger](https://docs.wpilib.org/en/stable/docs/software/telemetry/datalog.html). When I build the project, gradle fails because there was a segmentation fault running the tests.

The weird thing is that Google Test says the passed and the log file also gets created with the correct data. It gets a segmentation fault afterwards, so this would suggest that the problem is with the logger shutdown process.

**To Reproduce**
I created a repo of my code [here](https://github.com/Eaglestrike/wpilib-logger-bug). Clone this repo and build it. When gradle runs the Google Test, the build fails because there was a segmentation fault. To have a more detailed info about what happened use the command `./gradlew runFrcUserProgramTestLinuxx86-64ReleaseGoogleTestExe --info --scan --stacktrace`.

To see that the code builds successfully without the WPILib logger, the original Elevator Simulation example without the WPILib logger is on the [`elevatorsim-original-example`](https://github.com/Eaglestrike/wpilib-logger-bug/tree/elevatorsim-original-example) branch. FYI you can compare two branches with `git diff [other-branch-name]`.

**Expected behavior**
The gradle build should be successful without the logger causing a segmentation fault.

**Screenshots**
The build log is more applicable here, so I'm providing those:
- [Simple build log](https://github.com/Eaglestrike/wpilib-logger-bug/blob/main/build-run.txt)
- [Complex build log](https://github.com/Eaglestrike/wpilib-logger-bug/blob/main/detailed-google-test-run.txt)

**Desktop (please complete the following information):**
 - WPILib Version: [e.g. 2021.3.2]
 - OS: Arch Linux
 - GCC 12.2.1

**Additional information**
This is not specific to the elevator simulation example, I tried it in other environments and got the same results.

PeterJohnson, 2023-02-20:
I'm pretty sure this is a duplicate (or similar root cause) as #4625.

gaborszita, 2023-02-20:
@PeterJohnson Thank you for your quick response. Do you have an ETA for when this will be fixed?

PeterJohnson, 2023-02-20:
No, we don’t.  Note this is a destructor side issue, so the test itself works, it’s just crashing on exit.

gaborszita, 2023-02-20:
So you mean it's normal that the build is failing? I should create tests that pass but cause the build to fail? This would obviously be a very frustrating user experience for our team since everybody will be seeing build failed when in reality it passed, WPILib logger just crashed. It would also screw up our GitHub Actions check that builds the robot code and runs the tests.

PeterJohnson, 2023-02-20:
Yes, it’s a bug that needs to be fixed.

gaborszita, 2023-02-22:
Update:
1. This error is only present when networktables logging is enabled, so this would suggest that the problem is with networktables logging.
2. This error is only present on Linux, it builds and the test passes on Windows. Credit to @jkleiber for this.

gaborszita, 2023-02-22:
@PeterJohnson Do you know why this error is only present on Windows? Are you doing something differently with threads on different platforms?

gaborszita, 2023-02-22:
UPDATE: It seems I fixed the issue. I'll clean up some stuff and create a PR.

gaborszita, 2023-02-22:
PR closed for now as another issue has also discovered and I want to prevent somebody accidentally merging this half-solution. I'll continue with this tomorrow and reopen the PR if I find a solution.

PeterJohnson, 2023-02-22:
We aren’t doing anything different with threads on windows, but cross-module destruction order is not defined by C++, so different compilers/systems may destroy things in a different order.  The order can also be affected by things like the order of libraries passed to the linker.

gaborszita, 2023-02-22:
> PR closed for now as another issue has also discovered and I want to prevent somebody accidentally merging this half-solution. I'll continue with this tomorrow and reopen the PR if I find a solution.

Reopened. Ready to be merged.

PR #5125 (2023-02-24): [wpilibc] Fix DataLogManager crash on exit in sim
Closes #5120 

I added an Instance destructor that stops the network tables logging. This prevents the segmentation fault on exit.
closed 2023-02-24 by PeterJohnson