wpilibsuite/allwpilib #5124: WPILib 2023.4.1 Install Fails

Crossle86, 2023-02-22:
Downloaded install to Windows 10 system. Windows will not run the installer saying the file has a virus or potentially unwanted software. Tried several approaches to execute the installer but get the error each time. Never seen this before...

rzblue, 2023-02-22:
If this is the blue Windows Defender Smartscreen window, you can click "more info" to reveal a "Run Anyway" button.

Crossle86, 2023-02-22:
No not that window. If you just click on the installer, you get the first image. If you run as administrator you get the second image. If you try to copy the installer you get an error saying the same as second image.

[image]

[image]

Crossle86, 2023-02-22:
I  also tried downloading a second time, same thing happens.

sciencewhiz, 2023-02-22:
If you scan it with windows defender, what virus does it detect?

Crossle86, 2023-02-22:
Ok. So Avast security is blocking the installer (I forget it is active on this school district PC) and it says this:

[image]

Interesting...when I open the Avast quarantine it tells me the quarantine is empty.

So is there a problem with the download or is Avast misidentifying a problem and I should have Avast ignore the installer? I'm inclined to do that but just a bit hesitant.

sciencewhiz, 2023-02-22:
Since WPILIb 2023.4.1 has been out for days, assuming it was a real virus, presumably multiple people would have reported it by now.

You can report a false positive here: https://www.avast.com/false-positive-file-form.php#

gaborszita, 2023-02-22:
> Ok. So Avast security is blocking the installer (I forget it is active on this school district PC) and it says this:
>
> [image]
>
> Interesting...when I open the Avast quarantine it tells me the [...]

Avast identifies it as FileRepMalware, which according to [this](https://appuals.com/what-is-filerepmalware-and-should-you-remove-it/) source is given by Avast when all of the following conditions are met:
- The file is not added to the Antivirus clean chest
- The file is not signed by any publisher or the AV doesn’t trust the signature.
- The file isn’t prevalent enough – meaning that not enough users have tried to download, launch or use the file yet

WPILib 2023.4.1 is still relatively new, so the following conditions were probably met and it gave you the FileRepMalware. Don't hesitate to bypass it, antivirus software give countless false positives. If you know you downloaded it from a credible source, there's nothing to be afraid of.

Crossle86, 2023-02-22:
So it Avast is objecting because in a sense, it just does not recognize the installer file? Fun. Problem was the error messages looked like windows messages as opposed to a separate tool. In any case I created an Avast exception for the installer and installed successfully. Funny after so many WPILib installers,  Avast choked on this one.

Thanks for the great feedback.

Just a side note: with Avast installed, Windows Defender  is completely disabled. Can't check the file with it. Might be possible with tinkering but no time for that.

Crossle86, 2023-02-22:
Used VirusTotal and it supports the Avast false positive diagnosis.

closed 2023-02-22 by Crossle86