wpilibsuite/allwpilib #5126: Consider checking at runtime that the HID device internally used by a XboxCommandController is indeed a Xbox controller

gartaud, 2023-02-22:
This is a follow up to the issue described at https://www.chiefdelphi.com/t/commandxboxcontroller-lefttrigger-and-righttrigger-not-triggering/427504/38

If a XBoxCommandController is constructed by passing a port that corresponds to a Xbox controller that was not properly recognized as such by the NI Driver Station (either because it is not a Xbox controller, or because the NI Driver station somehow recognized it as a DInput device rather than a XInput device due to an alleged bug), then we periodically get the following somewhat cryptic message in the console output:

"Joystick axis 5 on port n not available, check if controller is plugged in", where n is the number of the port

The request is to provide a clearer message instead or in addition to the current message such as:

"Controller on port n is not recognized as a Xbox controller, Check if controller is plugged in, and that it is properly recognized as a XInput device by the Driver Station (it should show six axes numbered from 0 to 5: L X Axis, L Y Axis, L Trigger, R Trigger, R X Axis, R Y Axis). If you use a Logitech controller, make sure that the switch under the controller is set to X rather than D."

Thanks!

Starlight220, 2023-02-22:
Such a detailed message would be problematic, as it requires a lot more knowledge of the setup than the library can know.

How is the current message cryptic? It specifies very obviously what the problematic index and port is. _Why_ those indexes are problematic is a completely different question, and has a multitude of possible answers. The most common one by far is unplugged device, thus the message suggests that's the cause. If we had an HID troubleshooting doc we could link to it, but we don't have such a doc at the moment and I don't know what such a doc would include.

Adding a check in the Xbox classes that the underlying HID device is indeed an Xbox would be a good idea, but is rather limited by the fact that we don't have valid DS data when the object is created.

gartaud, 2023-02-22:
The library can access the info via DriverStation.getJoystickIsXbox(<port>) at runtime.

While the most common reason for this condition to happen prior to the 2023 season was for sure an unconnected controller, a common reason in 2023 is the controller not being properly recognized as a XInput device due to an alleged bug in the NI Driver Station.