wpilibsuite/allwpilib #5127: Consider implementing a virtual keyboard in Sim GUI that would behave like a XInput Xbox Controller

gartaud, 2023-02-22:
This is a follow up to https://www.chiefdelphi.com/t/commandxboxcontroller-lefttrigger-and-righttrigger-not-triggering/427504/37

And is related to https://github.com/wpilibsuite/allwpilib/issues/5126

As of today, it is not possible to fully simulate a proper (XInput) Xbox Controller in Sim GUI. The predefined virtual controller "keyboard 0" simulates a DInput device.

This request is for Sim GUI to provide a new virtual controller (e.g. keyboard 3) that would provide six axes that behave like a (XInput) Xbox Controller and would return true when DriverStation.getJoystickIsXbox() is called against the port it is assigne to.

Thanks!

Starlight220, 2023-02-22:
Perhaps more feasible would be a `IsXbox` checkbox for the keyboard-sourced sim HID devices?

gartaud, 2023-02-22:
Ideally would also provide six axes out of the box, and the proper travel range.