wpilibsuite/allwpilib #5134: Make `desaturateWheelSpeeds` an instance method

hadley31, 2023-02-24:
**Describe the bug**
https://github.com/wpilibsuite/allwpilib/blob/e5a619763354c9ad133399246dfda9b5693ddd20/wpimath/src/main/java/edu/wpi/first/math/kinematics/SwerveDriveKinematics.java#L263-L268

The parameter `currentChassisSpeed` above should really be the `desiredChassisSpeed`. If you pass the current chassis speeds, and the robot is stationary, this results in a 0 velocity for all swerve modules, preventing the robot from moving.

**To Reproduce**
Steps to reproduce the behavior:
1. Pass in `new ChassisSpeeds()` to `currentChassisSpeed`

**Expected behavior**
`desaturateWheelSpeeds` should return speeds based only on desired values, not values from feedback. Ideally this would be an instance method, and the speeds would be calculated based on `moduleStates` automatically, and not need to be passed in separately. Or possibly leave the static method, and add an instance method which calls the static method with the calculated chassis speeds.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.4.1
 - OS: Windows 10
 - Java version 17

calcmogul, 2023-02-24:
An instance method of what class?

hadley31, 2023-02-24:
> An instance method of what class?

The `SwerveDriveKinematics` class

calcmogul, 2023-02-24:
@JoshuaAN @r4stered thoughts on this issue since you're both familiar with what this code does?

hadley31, 2023-02-24:
I am thinking along the lines of how `toChassisSpeeds()` takes in an array `SwerveModuleState`. I'm honestly not sure why the class stores the module states, since they are only ever being set and never used. I would think for a given method call, `moduleStates` and the passed chassis speeds should be correlated? I'm suggesting something along the lines of:
```java
public void desaturateWheelSpeeds(
      SwerveModuleState[] moduleStates,
      double attainableMaxModuleSpeedMetersPerSecond,
      double attainableMaxTranslationalSpeedMetersPerSecond,
      double attainableMaxRotationalVelocityRadiansPerSecond) {
  desaturateWheelSpeeds(
    moduleStates,
    this.toChassisSpeeds(moduleStates),
    attainableMaxModuleSpeedMetersPerSecond,
    attainableMaxTranslationalSpeedMetersPerSecond,
    attainableMaxRotationalVelocityRadiansPerSecond);
}
```

r4stered, 2023-02-24:
Yeah the parameter name is confusing and should be renamed to desiredChassisSpeed or something like that. For example in our swerve code from this year. The parameters passed in are the desired speeds, not measured speeds. 

```
void str::SwerveDrivebase::Drive(
  units::meters_per_second_t xSpeed,
  units::meters_per_second_t ySpeed,
  units::radians_per_second_t rotSpeed,
  bool fieldRelative,
  bool openLoopDrive,
  bool voltageComp
) {
  auto states = kinematics.ToSwerveModuleStates(
    fieldRelative ? frc::ChassisSpeeds::FromFieldRelativeSpeeds(xSpeed, ySpeed, rotSpeed, imu.GetYaw()) :
                    frc::ChassisSpeeds(xSpeed, ySpeed, rotSpeed)
  );

  units::meters_per_second_t maxModuleSpeed = str::Units::ConvertAngularVelocityToLinearVelocity(
    str::motor_rpms::FALCON_MAX_RPM,
    str::swerve_physical_dims::DRIVE_WHEEL_DIAMETER / 2
  );

  kinematics.DesaturateWheelSpeeds(
    &states,
    frc::ChassisSpeeds(xSpeed, ySpeed, rotSpeed),
    maxModuleSpeed,
    str::swerve_drive_consts::MAX_CHASSIS_SPEED,
    str::swerve_drive_consts::MAX_CHASSIS_ROT_SPEED
  );
  ....
```

I agree that the class doesnt really need to hold any state of the current modules.

calcmogul, 2023-02-24:
The member variable is left over from when the kinematics reused an internal buffer and returned it from functions.

PR #5393 (2023-06-20): Rename currentChassisSpeed to desiredChassisSpeed
Fixes #5134
closed 2023-06-20 by PeterJohnson