wpilibsuite/allwpilib #5144: Add an `allowedSharedRequirement` parameter for parallel command compositions

Starlight220, 2023-02-26:
Currently, parallel command compositions ensure no shared requirements internally. This can be a problem for in-subsystem command factories.
We should have an optional arg to relax this restriction.

Open questions: should it be a single boolean parameter to completely remove the check, or accept the exact subsystem(s) that may be shared? If the latter, a single parameter or vararg? Do we require a subsystem that's passed as allowed for sharing but no composition member requires it?

prateekma, 2023-02-27:
Do you have an example where this is a problem currently?

Starlight220, 2023-02-27:
The shooter commands in the RapidReactCommandBot example involves two commands that require `this` subsystem composed in parallel. Situations like this are common.
The two solutions are splitting the subsystem to two individually-requirable elements, or allowing shared use of a requirement in a parallel composition.

TheTripleV, 2023-02-27:
I think adding support for nested subsystems would be better than an escape hatch from requirements checking.

Nested subsystems:
A subsystem can register subystems within it. In a command, requiring a subsystem requires the specified subsystem and all of it's children.

Ex: A telescoping arm could be structured in code as an arm subsystem containing a pivot subsystem and a telescope subsystem.