wpilibsuite/allwpilib #5150: Add `DeferredCommand` for commands that need runtime construction

Oblarg, 2023-02-28:
This would work similarly to `ProxyCommand` except it would not expose the wrapped command to the scheduler, making it more suitable for composition inside of single-subsystem command groups (`ProxyCommand` is designed for forking off to an unrelated requirement, and the semantics there are not well-suited to deferred execution inside of an ordinary composition).

Starlight220, 2023-02-28:
We had this problem with SelectCommand, which is why we moved the deferred functionality to ProxyCommand: when the command's construction is deferred, there's no way to figure out requirements etc. Hence, deferred are proxied.

Oblarg, 2023-02-28:
DeferredCommand would declare its requirements normally, and because it would wrap the supplied command internally it would sidestep the current problem ProxyCommand has with the proxied command's requirements possibly conflicting with those of the group it's composed in.

Starlight220, 2023-03-02:
That requires duplicate declaration of requirements, which is easy to miss.

Oblarg, 2023-03-02:
Not if we use subsystem factories to do the requirements implicitly.

PR #5566 (2023-10-27): [cmd] Add DeferredCommand
Closes #5150
Allows commands to be constructed at runtime without proxying.

Requirements must be specified explicitly, even if there are no requirements. Java uses a `Set<Subsystem>`. Not sure how to accomplish this in c++. The reason for this is that requirements must be known before runtime (and therefore before the contained command is constructed, and users will likely forget to include the proper requirements. 

The other potential tripping point is that the supplier *must* create a new command each time it's called, because this is considered a command composit [... cut]
closed 2023-10-27 by PeterJohnson