wpilibsuite/allwpilib #5151: Add `Trigger` factories for common robot state events

Oblarg, 2023-03-01:
These would sugar event bindings for transitions between the various robot control modes, etc:

```typescript
// naming and location TBD
static Trigger teleoperated() {
  return new Trigger(DriverStation::isDisabled);
}
```

The static factory methods doing this could live on `Trigger`, or on the classes whose state they're relevant to.

isaec, 2023-03-03:
This would certainly be nice, it can already be achieved by modifying the Robot.java file and calling new methods in your RobotContainer (is there a better way at present?). I figured there was some reason these events weren't exposed.

Starlight220, 2023-03-03:
> it can already be achieved by modifying the Robot.java file and calling new methods in your RobotContainer (is there a better way at present?)

It could be achieved like the implementation in the original post: `new Trigger(DriverStation::isAutonomous)` etc. 

If you're calling Robot from your RobotContainer, that's a circular dependency code smell.

PR #5902 (2023-11-14): [commands] Add static Trigger factories for robot mode changes
Included factories:

- [x] `autonomous()`
- [x] `teleop()`
- [x] `disabled()`
~~- [ ] `endOfMatch()` (would use `isTeleopDisabled()`)?~~

Are there any other conditions that should be included? I chose to use the `is[Mode]Enabled()` variants to ensure they only trigger when a match is in play. 

Resolves #5151 
Resolves #5916

Still needed:

- [x] Javadocs
- [x] C++ impl
- [x] C++ docs
closed 2023-11-14 by PeterJohnson