wpilibsuite/allwpilib #5152: Official Rust implementation

ultimatePurpleTruth, 2023-03-01:
Currently, the C++ programming language is slowly but surely becoming outdated, with memory-safe languages such as Rust gaining traction in recent years. Rust is a modern, safe, and efficient systems programming language, making it a perfect fit for WPILib and the FIRST Robotics Competition. An official Rust implementation of WPILib, or an official set of Rust bindings to WPILibC, would be tremendously helpful for new teams who wish to develop their robot programs using a lower-level systems language but do not wish to deal with the catastrophic runtime errors that C++ programs can sometimes cause.

I'm aware that this is not exactly a "feature request" in the traditional sense, and is quite non-trivial as it has been only C++ and Java for a really long time.

calcmogul, 2023-03-01:
Several unofficial attempts have been made before. The latest one I'm aware of is https://github.com/first-rust-competition/first-rust-competition. They usually fizzle out because WPILib development moves too fast, it's too much work to maintain, and the original project owners leave after they graduate.

Our active developers primarily know (modern) C++, and the Java port is typically copy-pasted C++ with small modifications since the syntax and APIs are so similar. Very few of us have Rust experience.

According to https://file.tavsys.net/wpilib/loc.pdf (generated from https://github.com/calcmogul/loc-grapher), we maintain about 320k lines of C++ code. Rewriting all of that in Rust would be a massive undertaking, and keeping it in sync with the rest of the library would be difficult. I don't know how linear algebra support would be handled either. The last time I checked, [nalgebra](https://www.nalgebra.org/) was missing a lot of the decompositions we use from [Eigen](https://gitlab.com/libeigen/eigen/), so a pure Rust port of wpimath wouldn't be possible.

It would be more tractible to take Robotpy's approach of autogenerating wrappers around the C++ version, although wrapping C++ wpimath is hard due to all the templates.

Starlight220, 2023-03-01:
First of all, there is a somewhat-abandoned community project: https://github.com/first-rust-competition. 

As for making it officially supported, I'd expect the main obstacles being its steep learning curve and lack of popularity. Rust is great and many other things, but one thing it's not is beginner friendly.

calcmogul, 2023-03-01:
> Rust is great and many other things, but one thing it's not is beginner friendly.

At least Rust has an official book, I suppose: https://doc.rust-lang.org/stable/book/

Placating the borrow checker is the main source of complaints I've seen from random new Rust users on the internet. I agree that broken code shouldn't compile in the first place, but I'm wondering how easy it is to write broken code compared to working code on the first attempt, and how easy the code is to fix based on the compilation errors.

virtuald, 2023-03-02:
Autogenerating wrappers around the C++ version of WPILib for python is a lot of work. I suspect it would be possible for someone *very* experienced with Rust and C++ to do a similar effort -- but it requires someone with both expertises and a lot of patience. It took about 2 months to really get the RobotPy bindings working -- but that was because pybind11 took care of a lot of the hard work for me already.

An "easier" approach that I've floated internally is to compile all of WPILibC to WebAssembly. If you could make a WASI interface to the native C HAL, in theory everything else could be compiled by clang into WASM. Once you have that WASM code, in theory you probably could export some API from the C++ code objects so that it could be accessed by other WASM code. There are lots of languages that can be compiled into WASM -- including Rust.

However, I'm not sure how workable that actually is, I've heard that a lot of Webassembly stuff is still mildly vaporware when it comes to native interfaces. It would be really interesting for someone to experiment with this and see how far you could get though.

closed 2023-08-02 by calcmogul