wpilibsuite/allwpilib #5153: [ntcore4] Since 2023.4.1, client is only receiving 1 value on a subscription to every topic

tsar-boomba, 2023-03-02:
**Describe the bug**
In both simulated and real code since 2023.4.1, [my client](https://github.com/tsar-boomba/network-tables-rs) only received one value even though my subscription is an empty string with the `all` property set to `true`.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the robot with version 2023.4.x
2. Connect a client with a subscription to `""` and `{"all": true}`
3. Don't receive all values

**Expected behavior**
Receive values of all topics the server recognizes.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.4.1 & 2023.4.2
 - OS: Mac OS
 - Java version 17
 - C++ version N/A

I am aware that my client had issues in the past, but I've switched to processing server message synchronously, and the behavior is as expected in 2023.3.2, so I do not believe that my client is the culprit this time.

PeterJohnson, 2023-03-02:
Did you subscribe with `prefix` set to true?

tsar-boomba, 2023-03-02:
I am using prefix true. I'll have to check and make sure I'm handing all messages in a frame. Thanks for the pointer, I'll report back with the results.

tsar-boomba, 2023-03-02:
That did seem to be the problem, thank you for your help once again.

closed 2023-03-02 by tsar-boomba