wpilibsuite/allwpilib #5166: [wpilibj/c request] Expand the RobotController class to provide system resource data

mjansen4857, 2023-03-07:
The driver station currently provides some RoboRIO system resource information, but there is currently no way to read this data programmatically for use in custom logging solutions or dashboards.

It would be nice if we could use the RobotController class to read certain resource usage data, including:
- CPU usage
- RAM usage
- Storage usage
- CAN utilization
- Processor temp

It would also potentially be nice to have access to network data (bandwidth, packet loss, etc)

virtuald, 2023-03-07:
It's just linux, why not get this information the normal way?

calcmogul, 2023-03-07:
What's the normal way?

virtuald, 2023-03-07:
All the information is in /proc or /sys somewhere.

calcmogul, 2023-03-07:
Here's a way to get CPU usage: https://stackoverflow.com/a/16736599

mjansen4857, 2023-03-07:
It looks like can utilization is already available through the RobotController.getCANStatus() method.

But, this request is for an easier way to get this info without having to parse through all of that stuff ourselves. There is access to this stuff at some lower level to send it to the driver station so it would be nice to pass that up further to the user. Basically, it is currently a lot of work to do this on a per-team basis just to include this data in a wpi datalog.

PeterJohnson, 2023-03-08:
I agree with adding utility methods here.  In general you can’t use shell commands like “free -m” because that requires forking, which can easily run out of memory; you have to pull the info directly from the /proc text files.

TandeJ, 2023-05-21:
Commenting here to bump.
Network use I think is especially important with NT growing pains and everyone starting to dump hundreds of fields over FMS.

Starlight220, 2023-09-27:
Network usage should be at `/proc/net/dev`, [(source)](https://man7.org/linux/man-pages/man5/proc.5.html)

Starlight220, 2023-09-30:
I have a proof of concept in my fork, will refine and PR in the upcoming days.