wpilibsuite/allwpilib #5176: protected vs private

truher, 2023-03-09:
hi, wpilib is a great teaching tool for my students.  it's helped them learn topics in control engineering and java programming that would have been hard to teach any other way.

it would be an even better teaching tool if it never used the word "private," but instead used "protected" everywhere.  several times this season we've wanted to modify the behavior of some library class, and we've resorted to cutting and pasting the whole thing.

the most recent example is the delay correction described below[1], which most naturally wants to be applied in SwerveControllerCommand or SwerveDriveKinematics, but there's no easy way to do it because all the state is private.

if everything were protected instead, the students could engage more deeply with the library, using it as a starting point for software experimentation.

obviously it would be ridiculous to maintain all internal state as a stable API, but I think if you made the contract clear ("expect breaking changes"), it would be a net win.

wdyt?

[1] https://www.chiefdelphi.com/t/field-relative-swervedrive-drift-even-with-simulated-perfect-modules/413892/15

auscompgeek, 2023-03-09:
Previous discussion: https://github.com/wpilibsuite/allwpilib/issues/1611

PeterJohnson, 2023-03-09:
That issue also got converted into a discussion (https://github.com/wpilibsuite/allwpilib/discussions/3237).  It might be best to simply continue the discussion there.

closed 2023-08-02 by calcmogul