wpilibsuite/allwpilib #5181: SimGUI match time does not match FMS behavior

jwbonner, 2023-03-11:
When setting the match time countdown using the SimGUI, the match time is reported to the robot code with sub-second precision. Based on our log files, the real DS and FMS seem to report the match time only to the nearest second. This could cause different behavior in sim and on the real robot, such as if a user checks whether the match time is equal to an integer value (which is probably not the best approach anyway but should behave the same in sim and on the real robot).

ThadHouse, 2023-03-11:
This is actually a place where the real ds actually has different behavior when it’s connected to the fms or not. When it’s hooked up to the fms, it counts down in integer increments. When not hooked up to the fms it’s a floating point value. But there’s also the fun quirk where when in normal teleop and autonomous modes, and not practice mode, it counts up rather then down. 

So there’s actually a lot of different behaviors here, even on a real robot. That’s actually why the documentation for that api says it’s not reliable. It’s different even on the official competition stuff.

jwbonner, 2023-03-11:
So if I understand correctly the simgui matches the behavior of the DS running in practice mode while not connected to the FMS? That's probably reasonable given the inconsistencies, though I think the documentation could be clearer. Noting the floating-point vs integer difference seems useful beyond just "The Practice Match function of the DS approximates the behavior seen on the field." To me that suggests that code tested on one will work on the other, which isn't always true. Noting what "approximates" means more specifically would make it easier to avoid those issues.

PR #5232 (2023-08-04): Update GetMatchTime docs and units
Closes #5181
closed 2023-08-04 by PeterJohnson