wpilibsuite/allwpilib #5184: Install directory for the VS Code and tools

calculatorboy, 2023-03-11:
My laptop has very little storage left on its c drive, so I have to move wpilib over to my second drive on my laptop. This comes with a lot of issues and takes a lot of time. When I do this the extensions don't stay installed. To fix the extensions not being installed I have to add each one and reload vs code. WPILib extension takes many reloads to work.

This could be easily fixed by allowing a user to specify an install directory which would fix this issue easily.

I have been copying and pasting the entire wpilib vs code folder onto my second drive

ThadHouse, 2023-03-12:
Moving that folder actually breaks a lot of things, since that folder is hardcoded across the entire ecosystem. It breaks the gradlerio offline cache, launching tools from both the ds and vs code, and vscodes lookup of the jdk. We don’t currently have any plans to make all of those things configurable, which means that this wouldn’t be feasible either.

PeterJohnson, 2023-03-12:
You can use directory symlinks (mklink /D) to set up the C:\Users\Public\wpilib folder to point to another drive. You'll also want to point C:\Users\<yourname>\.gradle to the other drive.

closed 2023-08-02 by calcmogul