wpilibsuite/allwpilib #5189: Consider automatically including information about source code used to build user binaries and provide easy way to retrieve it at runtime

gartaud, 2023-03-15:
This is a follow up to https://www.chiefdelphi.com/t/is-there-a-way-to-send-automatically-generated-software-build-and-or-deployment-timestamp-to-smartdash/429965

Description of problem duplicated hereunder:
"Sometimes when programmers juggle between connecting to the Robot’s radio to test code changes and connecting to the school 's Wi-fi to pull and/or push code changes, they forget what they have last deployed to the robot. They can always redeploy, but that takes a few minutes, and when done repeatedly over an afternoon, can end up wasting quite a bit of time.

It would therefore be neat if one could easily tell by looking at the smartdash/shuffleboard when the code running on the robot was built and/or deployed (using the time reference from the laptop used to build and/or deploy being good enough)."

Ideally the solution would work independently of Git, but since most FRC teams use (or should use) Git, relying on Git would be totally acceptable.

Useful information, as suggested by others, could include: 
build date
Git and/or Windows username

last Git commit hash
last Git commit message
if source code was modified since last commit

Ideally the information would be automatically collected during the build (e.g. by expanding on the current Gradle file), and an easy way to access it at runtime would be provided by WPILib.

Thanks!

calcmogul, 2023-03-15:
https://docs.wpilib.org/en/stable/docs/software/advanced-gradlerio/deploy-git-data.html

chauser, 2023-03-16:
The one thing I've noticed where some more direct support would be helpful is in importing one year's robot project to the next year's WPILib. The code for collecting and saving git version information in `build.gradle` gets removed. An annoyance rather than a showstopper.

calcmogul, 2023-03-16:
Since user build.gradle modifications can be so wide-ranging, I think the only thing we could do is generate a patch from the 2022 default build.gradle to the project build.gradle, then try applying it to the 2023 default build.gradle.

ThadHouse, 2023-03-16:
I actually kind of have a plan for that. We won’t make any guarantees things won’t break, but it should allow for simple helpers like that to be kept year over year.

gartaud, 2023-03-16:
> We already have docs for how to do this: https://docs.wpilib.org/en/stable/docs/software/advanced-gradlerio/deploy-git-data.html
> 
> The robot code could read the deployed files and publish their contents over NT.

Thanks. The request is that the Gradle file should come preset for this out of the box, and that the parsing of the deployed file(s) also be built-in in WPILib.

Admittedly this seems quite lazy on my part, but most teams would benefit from such a mechanism, so it makes sense to have it built-in, and not under "Advanced GradleRIO" which sounds a bit scary ;-)

Traceability between source code and binaries is important, and once implemented, this mechanism could be highlighted in "Basic Programming", for example just after https://docs.wpilib.org/en/stable/docs/software/basic-programming/git-getting-started.html

auscompgeek, 2023-03-16:
FWIW if this becomes baked into WPILib, we should consider having the same information that the RobotPy deploy bundles up too. https://robotpy.readthedocs.io/en/stable/dev/deploy.html#deploy-artifacts

closed 2026-02-21 by gartaud