wpilibsuite/allwpilib #5190: Multi-constant case labels

NanoDecade, 2023-03-16:
I am currently trying to use a multi-constant case label in Java. I get an error from vscode saying that multi-constant case labels are available from Java 14 and onwards. Is wpilib 2023 not supposed to run java 17?

Its confusing because vscode knows which versions of java will work but it will not run or build. 

This is image that it shows. 
[image]

ThadHouse, 2023-03-16:
We run on Java 17, but the project is set to 11 as an escape hatch for us if we needed to revert in the middle of the season. If you want to use 17, you can do so removing the lines in build.gradle that force the project to 11.

You’ll likely then need to run the Clean Java Workspace command in vscode to force a full intelligence reset.

NanoDecade, 2023-03-16:
So like this? 
[image]

Also how do I run the Clean Java Workspace command?

ThadHouse, 2023-03-16:
Clean Java workspace is a command in vscode. You can open the command palette and search for it.

NanoDecade, 2023-03-16:
Nevermind, turns out cleaning was unnecessary. Thanks for you help, it is working now!

closed 2023-03-16 by NanoDecade