wpilibsuite/allwpilib #5205: Deduplicate common functionality in pose estimators

calcmogul, 2023-03-19:
The odometry classes are wrappers around a kinematics object that converts from wheel movements to chassis movements, and a call to `Pose2d.exp()`. The pose estimators are wrappers around an odometry object, measurement latency compensation, and a Kalman filter for fusing pose measurements from computer vision that correctly represents the probability distribution in the Lie group SE(2).

The number of odometry and pose estimator classes is getting unwieldy (3 odometry in C++, 3 pose estimator in C++, 3 odometry in Java, 3 pose estimator in Java).

There's enough common complexity in the pose estimator classes to refactor into a RobotPoseEstimator class, which would make the pose estimators easier to maintain. The odometry classes are pretty simple as it is though, and the common functionality is already refactored into `Pose2d.exp()`.

srimanachanta, 2023-04-26:
How do you think we should handle the update call? We can create a `RobotPoseEstimator` that combined vision data with an update method that handles the `Pose2d::exp` call but then what will handle the rest of the math required for odometry? Although each odometry system is very similar, each one does things slightly differently (such as checking previous distances for tank drive).

calcmogul, 2023-04-27:
The pose estimators will still need wrapper classes for each drivetrain that handle the drivetrain-specific kinematics, but the common logic for the Kalman filter can be refactored out in the same way the odometry classes have the pose update refactored into Pose2d::exp().

bruingineer, 2023-04-28:
here is a very crude initial attempt at just taking out the filter matrices
[my forked branch](https://github.com/wpilibsuite/allwpilib/compare/main...bruingineer:allwpilib:Deduplicate-common-functionality-in-pose-estimators?diff=split)

all the pose estimators have an odometry object, but the function signatures are all different. So I'm not sure how to correctly consolidate that.

closed 2023-06-20 by PeterJohnson