wpilibsuite/allwpilib #5239: Joystick transitions are missed in DataLogManager log files

chauser, 2023-03-24:
**Describe the bug**
DataLogManager log files have incorrect button and POV information for the DriverStation much of the time. The problem also affects axis information but to a lesser extent because axes change much more frequently than buttons and POVs.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable data logging for the driverstation: `DataLogManager.start(); DriverStation.startDataLog(DataLogManager.getLog());`
2. In simulation or on the real robot, push buttons, POVs, and move axes.
3. Inspect log for missed transitions -- we find that in match data the button/POV states are frequently wrong. I've attached a log file from a simulation in which I moved the sticks and pressed all the buttons and POVs; it appears that all of the transitions were captured (eventually) except the last one for the POV which remained at 315 and the last one for button 10 (index 9) which remained pressed to the end although I had released them. Some of the transitions seem significantly delayed from how long I remember holding the buttons.

**Expected behavior**
When joystick state changes it is logged.

**Files**
The attached `.log` file is actually a `.wpilog` file but it can't be uploaded with that suffix.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.4.2
 - OS: Roborio
 - Java version: what is installed by WPILib
 - C++ version: N/A 
**Additional context**
I believe that the problem is caused by the tests used in DriverStation.JoystickLogSender.send method to determine whether or not to log the current state. If I've understood the code correctly, the comparisons look to see if the data for each kind of joystick data _passed to robot code_ has changed since the they were last copied from the cached versions. However, those data are updated at the frequency of the robot period. The DSLogging thread runs at only 10 Hz so transitions that were already picked up are no longer seen as having transitions. I think the correct comparison would be against what was previously logged.
[FRC_20230324_164532.log](https://github.com/wpilibsuite/allwpilib/files/11066062/FRC_20230324_164532.log)

chauser, 2023-03-25:
I tested a version of DriverStation.java in which I compared against the previous  logged values. The log tracked the joystick state changes correctly. 
[DriverStation.zip](https://github.com/wpilibsuite/allwpilib/files/11068099/DriverStation.zip)

PeterJohnson, 2023-03-25:
Thanks!  I’ll review and open a PR (or you can do it).

chauser, 2023-03-25:
I'll leave the PR to you. I took a look at the C++ code and it appears to already compare against the previous logged value, so no worries about that.

PR #5240 (2023-03-26): [wpilibj] DriverStation: Fix joystick data logs
Was logging relative to cached value rather than previously logged value.  Was also logging cached value instead of current loop value.  C++ implementation is correct.

Fixes #5239.
closed 2023-03-26 by PeterJohnson