wpilibsuite/allwpilib #5253: USing putData on a CTRE PDP throws an exception

blu28, 2023-04-05:
**Describe the bug**
If you instantiate a powerDistribution object (which implements Sendable) with a physical CTRE PDP and then use Smartdashboard.putData to put it on the dashboard you get this stack trace:

CAN: Message not found",
frc.robot.Main.main(Main.java:23)
edu.wpi.first.hal.PowerDistributionJNI.getTotalCurrent(Native Method)
edu.wpi.first.wpilibj.PowerDistribution.lambda$initSendable$2(PowerDistribution.java:211)
edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.lambda$addDoubleProperty$6(SendableBuilderImpl.java:398)
edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl$Property.update(SendableBuilderImpl.java:88)
edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.update(SendableBuilderImpl.java:180)
edu.wpi.first.wpilibj.shuffleboard.ComplexWidget.buildInto(ComplexWidget.java:33)
edu.wpi.first.wpilibj.shuffleboard.ShuffleboardLayout.buildInto(ShuffleboardLayout.java:132)
edu.wpi.first.wpilibj.shuffleboard.ShuffleboardTab.buildInto(ShuffleboardTab.java:145)
edu.wpi.first.wpilibj.shuffleboard.ShuffleboardInstance.update(ShuffleboardInstance.java:59)
edu.wpi.first.wpilibj.shuffleboard.Shuffleboard.update(Shuffleboard.java:67)
edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:363)
edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:130)
edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:349)
edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:442)
frc.robot.Main.main(Main.java:23)

**To Reproduce**
1. connect a CTRE PDP to the RoboRio via CAN.
2. Run a robot that instantiates a PowerDistribution object.
3. Put that object on the dashboard with putData.

**Expected behavior**
You get a widget on the dashboard with the PDP data.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - WPILib Version:2023.4.3
 - OS: Windows 10
 - Java version 17.0.5
 - C++ version NA

sciencewhiz, 2023-04-05:
What is the firmware version of the PDP?

blu28, 2023-04-06:
1.40.

rzblue, 2024-07-10:
This shouldn't be an exception, just a warning w/ stacktrace in the DS console

AustinShalit, 2026-11-13:
Closed by #7773 which refactored this interface. Please reopen if the new interface also has this issue.

closed 2026-11-13 by AustinShalit