wpilibsuite/allwpilib #5254: Send robot loop time data in DS packet and store in the DS log

virtuald, 2023-04-06:
**Is your feature request related to a problem? Please describe.**
Teams don't really know what their loop times are, and we have the watchdog now... but really, people ignore that.

**Describe the solution you'd like**
Send robot loop time data in DS packet and store in the DS log. Maybe the DS warns the user if it's too high with some widget when they're driving.

PeterJohnson, 2023-04-06:
And similar to the old control system, if loop times are over 20 ms for X amount of time, light up the radio LED.

PeterJohnson, 2023-04-06:
If we do this, and provide a different way to collect/examine the details, we could remove the print message entirely, which would decrease log noise quite a bit.

virtuald, 2023-04-06:
Could also only create log noise when FMS isn't connected, which would lead to the weirdest CD posts ever:

> URGENT: My robot only lags when it's NOT connected to FMS